dreadnought/python-daly-bms #1: Error - failed after 5 tries

SamLue, 2021-05-18:
Hey,

first of all: Thanks for your work and this Projekt.

And maybe this Error is only a Problem of my Setup.

I have a Daly Smart BMS 4S 12V 100A.
This BMS is connected via the Monitor Port and the original cable with an Lubuntu (Ubuntu 18.04.5 LTS). (With this cable I can read the data and configure the BMS through the sinowealth software from a windows PC)

samuel@nb-lubuntu:~$ python3 --version
Python 3.6.9

samuel@nb-lubuntu:~$ pip3 show pyserial
Name: pyserial
Version: 3.5
Summary: Python Serial Port Extension
Home-page: https://github.com/pyserial/pyserial
Author: Chris Liechti
Author-email: cliechti@gmx.net
License: BSD
Location: /home/samuel/.local/lib/python3.6/site-packages
Requires:

I have also set the right permission to the user (Added the user to the group dialout)

But I still get no data from my BMS. Instead I get the following Error:
samuel@nb-lubuntu:~$ daly-bms-cli  -d /dev/ttyUSB0 --soc
ERROR    [daly_bms.py:88] 94 failed after 5 tries
ERROR    [daly_bms.py:88] 90 failed after 5 tries
false

Which BMS do you use and how is it connected? Do you connect your BMS with UART-Port? At the moment it is used by the Bluetooth-Dongle and I have no USB-Calbe for UART.

I hope you can help me with my problems.
Sam

SamLue, 2021-05-18:
Here further information from my serial USB cable.
samuel@nb-lubuntu:~$ ls -la /dev/serial/by-id/
insgesamt 0
drwxr-xr-x 2 root root 60 Mai 18 14:11 .
drwxr-xr-x 4 root root 80 Mai 18 14:11 ..
lrwxrwxrwx 1 root root 13 Mai 18 14:11 usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0

maybe interesting?

dreadnought, 2021-05-18:
I'm using the 14S model from here: https://aliexpress.com/item/1005002205134036.html

It has a RS485 and UART port and has the USB adapter is connected to RS485, the Bluetooth module to UART. Both can be connected and used in parallel.

Please run the CLI command again with `--verbose` at the end. Then it will print out all messages send to and received from the BMS. The output could help me understand where the problem is.

I've also added the new parameter `--uart` now, which you could give a try. For that you have to pull the latest code from Github and run `./bin/daly-bms-cli` in it.

SamLue, 2021-05-19:
Supposedly the RS485 is not present on the 4s 100A. There is a connector to the right of UART, but it is not labeled. Therefore you should use the monitor port with the right USB-Cable for the computer.

Here the output with --verbose Switch
samuel@nb-lubuntu:~/daly$ daly-bms-cli -d /dev/ttyUSB0 --soc --verbose
DEBUG    [daly_bms.py:93] -- 94 ------------------------
DEBUG    [daly_bms.py:65] w a5409408000000000000000081
DEBUG    [daly_bms.py:110] 0 empty response for command 94
DEBUG    [daly_bms.py:83] 1. try failed, retrying...
DEBUG    [daly_bms.py:93] -- 94 ------------------------
DEBUG    [daly_bms.py:65] w a5409408000000000000000081
DEBUG    [daly_bms.py:110] 0 empty response for command 94
DEBUG    [daly_bms.py:83] 2. try failed, retrying...
DEBUG    [daly_bms.py:93] -- 94 ------------------------
DEBUG    [daly_bms.py:65] w a5409408000000000000000081
DEBUG    [daly_bms.py:110] 0 empty response for command 94
DEBUG    [daly_bms.py:83] 3. try failed, retrying...
DEBUG    [daly_bms.py:93] -- 94 ------------------------
DEBUG    [daly_bms.py:65] w a5409408000000000000000081
DEBUG    [daly_bms.py:110] 0 empty response for command 94
DEBUG    [daly_bms.py:83] 4. try failed, retrying...
DEBUG    [daly_bms.py:93] -- 94 ------------------------
DEBUG    [daly_bms.py:65] w a5409408000000000000000081
DEBUG    [daly_bms.py:110] 0 empty response for command 94
DEBUG    [daly_bms.py:83] 5. try failed, retrying...
ERROR    [daly_bms.py:88] 94 failed after 5 tries
DEBUG    [daly_bms.py:93] -- 90 ------------------------
DEBUG    [daly_bms.py:65] w a540900800000000000000007d
DEBUG    [daly_bms.py:110] 0 empty response for command 90
DEBUG    [daly_bms.py:83] 1. try failed, retrying...
DEBUG    [daly_bms.py:93] -- 90 ------------------------
DEBUG    [daly_bms.py:65] w a540900800000000000000007d
DEBUG    [daly_bms.py:110] 0 empty response for command 90
DEBUG    [daly_bms.py:83] 2. try failed, retrying...
DEBUG    [daly_bms.py:93] -- 90 ------------------------
DEBUG    [daly_bms.py:65] w a540900800000000000000007d
DEBUG    [daly_bms.py:110] 0 empty response for command 90
DEBUG    [daly_bms.py:83] 3. try failed, retrying...
DEBUG    [daly_bms.py:93] -- 90 ------------------------
DEBUG    [daly_bms.py:65] w a540900800000000000000007d
DEBUG    [daly_bms.py:110] 0 empty response for command 90
DEBUG    [daly_bms.py:83] 4. try failed, retrying...
DEBUG    [daly_bms.py:93] -- 90 ------------------------
DEBUG    [daly_bms.py:65] w a540900800000000000000007d
DEBUG    [daly_bms.py:110] 0 empty response for command 90
DEBUG    [daly_bms.py:83] 5. try failed, retrying...
ERROR    [daly_bms.py:88] 90 failed after 5 tries

And the same with the latest code from github an --uart Switch.

dreadnought, 2021-05-19:
The output shows that the BMS doesn't answer at all. 

You've mentioned that you've successfully connected to the BMS from Windows. Did you try it with the the Daly BMS Monitor software? https://www.dalyelec.cn/newsshow.php?cid=25&id=77&lang=1

They list two different software packages there, for my BMS I took the BmsMonitor. Would it be the same for your device model?

In the beginning I just run that software and captured the commands that it sends via a serial monitor software. So when you can also connect with the Daly software, you could also have a look at the data that gets send and received via the serial interface. I've used the trial of Serial Port Monitor for that: https://www.hhdsoftware.com/serial-port-monitor

SamLue, 2021-05-20:
OK,
I checked the BMS Monitor. But this Software doesn't work with my BMS on the Monitor Port.

The 2. Software of your Link (BMSTool - Sinowealth) is working with my BMS on this Port.

I captured the data with your serial port monitor and choosed "Text Exporter" as processing. So here are the first 1000 Lines (maybe 2-3 Seconds of scanning all bms data by the sinowealth software) https://pastebin.com/KxRq07iq

What do you think. Should I try to order a cable for UART Port? But then I can't use the Bluetooth-App parallel.

dreadnought, 2021-05-22:
Your captured data was helpful, it shows me that this BMS uses completely different different protocol. So it doesn't matter which cable you're using.

The file is quite long, but the interesting lines are starting with a tab character (`grep $'^\t' KxRq07iq.txt` on Linux).

First the software sends a command twice, e.g. `0a0102` for the first cell voltage, and then it receives a response like `0d03fd`, which I would interpret as 3.331V. Another example would be the total voltage, where `0a0b02` is the command and `340d0f` (13.325V) the answer. Summing the cell voltages up matches the total voltage quite close.

The command structure is not hard to understand:

- Byte 1: Always `0a`.
- Byte 2: A number from `01` to `18` (hex), which means that there are 24 different commands and they are most likely order the way that they are ordered in the Windows app.
- Byte 3: Always `02`.

The responses have 3 or 5 bytes, the first 2 or 4 bytes are the value, followed by 1 byte that I can't interpret. Maybe it's some kind of checksum or could tell us from which command this response is. 

Parsing voltages with Python can look like this:
```
>>> import struct
>>> struct.unpack('>H x', b"\x0d\x03\xfd")[0] / 1000
3.331
```

You don't have to understand everything above, I just needed a place to document what I found out. I don't know which values are actually interesting for you. Just getting the voltages seems to be easy, assuming that my results are correct so far.

The next part would be to go over all values in the Windows software, checking them one by one and creating a list of the name & value show in the GUI and the command & response in hex format from serial port monitor. The "Packet View" works better than text export in this case, because you can see the data in real-time and press DEL to clear the screen, before going to the next command.

With that information I should be able to implement most of it, even when I don't have such a device myself. But I can't promise anything of course.

scoleri, 2021-05-25:
Same issues for me.

What did work:
`>>> struct.unpack('>H x', b"\x0d\x00\xfd")[0] / 1000
3.328
>>> struct.unpack('>H x', b"\x0d\x01\xfd")[0] / 1000
3.329
>>> struct.unpack('>H x', b"\x0d\x02\xfd")[0] / 1000
3.33
>>> struct.unpack('>H x', b"\x0d\x03\xfd")[0] / 1000
3.331`

scoleri, 2021-05-25:
Does this document help?

[https://diysolarforum.com/resources/daly-smart-bms-manual-and-documentation.48/version/63/download?file=13550](url)

dreadnought, 2021-05-26:
I've pushed some code for reading the voltage values to a branch called `sinowealth`, please give it a try and post the output here.

You can get and run it like this:
```
git pull
git checkout sinowealth
python3 setup.py install
daly-bms-cli  -d /dev/ttyUSB0 --verbose --sinowealth --soc
daly-bms-cli  -d /dev/ttyUSB0 --verbose --sinowealth --cell-voltages
```
It's not finished yet, but it should show if the BMS responds.

> Does this document help?
> 
> [https://diysolarforum.com/resources/daly-smart-bms-manual-and-documentation.48/version/63/download?file=13550](url)

Unfortunately not, it describes the protocol that is already implemented here. It's not about the protocol that the Sinowealth app uses.

scoleri, 2021-05-26:
Looks better AWESOME - thanks!
```
root@battery:~# daly-bms-cli  -d /dev/ttyUSB0 --verbose --sinowealth --soc
DEBUG    [daly_sinowealth.py:38] message: bytearray(b'\n\r\x02'), 0a0d02
DEBUG    [daly_sinowealth.py:59] 0aab2f 3
{
  "total_voltage": 2.731
}
root@battery:~# daly-bms-cli  -d /dev/ttyUSB0 --verbose --sinowealth --cell-voltages
DEBUG    [daly_sinowealth.py:38] message: bytearray(b'\n\x01\x02'), 0a0102
DEBUG    [daly_sinowealth.py:59] 0cfd1c 3
DEBUG    [daly_sinowealth.py:38] message: bytearray(b'\n\x02\x02'), 0a0202
DEBUG    [daly_sinowealth.py:59] 0cfe2f 3
DEBUG    [daly_sinowealth.py:38] message: bytearray(b'\n\x03\x02'), 0a0302
DEBUG    [daly_sinowealth.py:59] 0cfd30 3
DEBUG    [daly_sinowealth.py:38] message: bytearray(b'\n\x04\x02'), 0a0402
DEBUG    [daly_sinowealth.py:59] 0cf663 3
DEBUG    [daly_sinowealth.py:38] message: bytearray(b'\n\x05\x02'), 0a0502
DEBUG    [daly_sinowealth.py:59] 000045 3
{
  "1": 3.325,
  "2": 3.326,
  "3": 3.325,
  "4": 3.318
}

dreadnought, 2021-05-26:
Thanks for testing it, great that it is already returning data. `total_voltage` wasn't correct, but should be now. Please update git, run it again, check if the values are equal to you see in the GUI. Right now I'm just dividing everything by 1000, I don't know if that works for all fields.

The `--temperatures` should also give a result now:
```
daly-bms-cli  -d /dev/ttyUSB0 --verbose --sinowealth --temperatures
```

dreadnought, 2021-05-26:
I guess your BMS is not 2°C cold, so let's divide temperatures only by 100, which results in 29°C. `--soc` should also be fixed now.

SamLue, 2021-05-26:
Yeah, that sounds greate!

But the new Branch doesn't work on my system. Do you have an idea what the problem could be?
samuel@nb-lubuntu:~/daly/python-daly-bms$ daly-bms-cli  -d /dev/ttyUSB0 --verbose --sinowealth --cell-voltages
usage: daly-bms-cli [-h] -d DEVICE [--status] [--soc] [--mosfet]
                    [--cell-voltages] [--temperatures] [--balancing]
                    [--errors] [--all] [--check] [--retry RETRY] [--verbose]
daly-bms-cli: error: unrecognized arguments: --sinowealth

samuel@nb-lubuntu:~/daly/python-daly-bms$ git status
Auf Branch sinowealth
Ihr Branch ist auf demselben Stand wie 'origin/sinowealth'.

nichts zu committen, Arbeitsverzeichnis unverändert
samuel@nb-lubuntu:~/daly/python-daly-bms$

Do you need more Information from the serial port monitor with the "Packet View"-Mode?
A few days are still left in the eval. (until 03.06.)

dreadnought, 2021-05-26:
>  daly-bms-cli: error: unrecognized arguments: --sinowealth

That shows that the most recent version is not installed. Please run `python3 setup.py install` again and retry.

> Do you need more Information from the serial port monitor with the "Packet View"-Mode?

Not for know, since the script can show the response from the BMS already, I can use the output from it. But we have to see if the output look similar to what the GUI shows. So please let me know if something doesn't look right.

SamLue, 2021-05-26:
I already have run python3 setup.py install

samuel@nb-lubuntu:~/daly/python-daly-bms$ sudo python3 setup.py install
/usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
running install
running bdist_egg
running egg_info
writing dalybms.egg-info/PKG-INFO
writing dependency_links to dalybms.egg-info/dependency_links.txt
writing top-level names to dalybms.egg-info/top_level.txt
reading manifest file 'dalybms.egg-info/SOURCES.txt'
writing manifest file 'dalybms.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-i686/egg
running install_lib
running build_py
creating build/bdist.linux-i686/egg
creating build/bdist.linux-i686/egg/dalybms
copying build/lib/dalybms/__init__.py -> build/bdist.linux-i686/egg/dalybms
copying build/lib/dalybms/daly_bms_bluetooth.py -> build/bdist.linux-i686/egg/dalybms
copying build/lib/dalybms/daly_bms.py -> build/bdist.linux-i686/egg/dalybms
copying build/lib/dalybms/error_codes.py -> build/bdist.linux-i686/egg/dalybms
copying build/lib/dalybms/daly_sinowealth.py -> build/bdist.linux-i686/egg/dalybms
byte-compiling build/bdist.linux-i686/egg/dalybms/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-i686/egg/dalybms/daly_bms_bluetooth.py to daly_bms_bluetooth.cpython-36.pyc
byte-compiling build/bdist.linux-i686/egg/dalybms/daly_bms.py to daly_bms.cpython-36.pyc
byte-compiling build/bdist.linux-i686/egg/dalybms/error_codes.py to error_codes.cpython-36.pyc
byte-compiling build/bdist.linux-i686/egg/dalybms/daly_sinowealth.py to daly_sinowealth.cpython-36.pyc
creating build/bdist.linux-i686/egg/EGG-INFO
installing scripts to build/bdist.linux-i686/egg/EGG-INFO/scripts
running install_scripts
running build_scripts
creating build/bdist.linux-i686/egg/EGG-INFO/scripts
copying build/scripts-3.6/daly-bms-cli -> build/bdist.linux-i686/egg/EGG-INFO/scripts
changing mode of build/bdist.linux-i686/egg/EGG-INFO/scripts/daly-bms-cli to 755
copying dalybms.egg-info/PKG-INFO -> build/bdist.linux-i686/egg/EGG-INFO
copying dalybms.egg-info/SOURCES.txt -> build/bdist.linux-i686/egg/EGG-INFO
copying dalybms.egg-info/dependency_links.txt -> build/bdist.linux-i686/egg/EGG-INFO
copying dalybms.egg-info/top_level.txt -> build/bdist.linux-i686/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating 'dist/dalybms-0.1.0-py3.6.egg' and adding 'build/bdist.linux-i686/egg' to it
removing 'build/bdist.linux-i686/egg' (and everything under it)
Processing dalybms-0.1.0-py3.6.egg
Removing /usr/local/lib/python3.6/dist-packages/dalybms-0.1.0-py3.6.egg
Copying dalybms-0.1.0-py3.6.egg to /usr/local/lib/python3.6/dist-packages
dalybms 0.1.0 is already the active version in easy-install.pth
Installing daly-bms-cli script to /usr/local/bin

Installed /usr/local/lib/python3.6/dist-packages/dalybms-0.1.0-py3.6.egg
Processing dependencies for dalybms==0.1.0
Finished processing dependencies for dalybms==0.1.0

But still:
samuel@nb-lubuntu:~/daly/python-daly-bms$ daly-bms-cli  -d /dev/ttyUSB0 --verbose --sinowealth --cell-voltages
usage: daly-bms-cli [-h] -d DEVICE [--status] [--soc] [--mosfet]
                    [--cell-voltages] [--temperatures] [--balancing]
                    [--errors] [--all] [--check] [--retry RETRY] [--verbose]
daly-bms-cli: error: unrecognized arguments: --sinowealth

scoleri, 2021-05-27:
Yes the numbers match.  I'd like to see amps if you can figure that out.  Thanks a ton.
```
✔ ~/python-daly-bms [sinowealth|⚑ 1]
20:55 # daly-bms-cli  -d /dev/ttyUSB0 --sinowealth --soc --cell-voltages --temperatures
{
  "total_voltage": 13.623,
  "soc_percent": 97.0
}
{
  "1": 3.426,
  "2": 3.429,
  "3": 3.43,
  "4": 3.34
}
{
  "external1": 30.0,
  "external2": 27.31,
  "ic1": 30.17,
  "ic2": 0.71
}
```

scoleri, 2021-05-27:
> I already have run python3 setup.py install
>
> samuel@nb-lubuntu:~/daly/python-daly-bms$ sudo python3 setup.py install
> /usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution [...]

You're probably not in the right branch.
```
cd python-daly-bms
git checkout sinowealth

```

SamLue, 2021-05-27:
No, I'm in the sinowealth branch.

samuel@nb-lubuntu:~/daly/python-daly-bms$ git status
Auf Branch sinowealth
Ihr Branch ist auf demselben Stand wie 'origin/sinowealth'.

nichts zu committen, Arbeitsverzeichnis unverändert
samuel@nb-lubuntu:~/daly/python-daly-bms$

What am I doing wrong?

scoleri, 2021-05-27:
Did you pull???  You're not getting the new code for whatever reason.

```
✔ ~/python-daly-bms [sinowealth|⚑ 1]
06:17 # git pull
Already up to date.

> No, I'm in the sinowealth branch.
> 
> samuel@nb-lubuntu:~/daly/python-daly-bms$ git status
> Auf Branch sinowealth
> Ihr Branch ist auf demselben Stand wie 'origin/sinowealth'.
> 
> nichts zu committen, Arbeitsverzeichnis unverändert
> samuel@nb-lubuntu:~/daly/python-daly-bms$
> 
> What am I doing wrong?

SamLue, 2021-05-27:
Yes, of course! The code is up2date.

Now I remove manually the files
/usr/local/bin/daly-bms-cli
/usr/local/lib/python3.6/dist-packages/dalybms-0.1.0-py3.6.egg

reinstalled it with:
`sudo python3 setup.py install`

But still the same.
daly-bms-cli: error: unrecognized arguments: --sinowealth

In the Scriptfile /daly/python-daly-bms/build/scripts-3.6/daly-bms-cli there is the line
`parser.add_argument("--sinowealth", help="BMS with Sinowealth chip", action="store_true")`
Strange!
Any futher ideas?

scoleri, 2021-05-27:
Is your path right?   Maybe you're using another installation ~/.local/bin?  something like that?  No idea.

SamLue, 2021-05-27:
> 
> 
> Is your path right? Maybe you're using another installation ~/.local/bin? something like that? No idea.

Yeah! Thanks man. There was also a installation in ~/.local /bin

But still no answer from the bms.
samuel@nb-lubuntu:~$ daly-bms-cli -d /dev/ttyUSB0 --sinowealth --soc --verbose
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0b\x02'), 0a0b02
DEBUG    [daly_sinowealth.py:83] empty response for command b
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x10\x04'), 0a1004
DEBUG    [daly_sinowealth.py:83] empty response for command 10
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x13\x02'), 0a1302
DEBUG    [daly_sinowealth.py:83] empty response for command 13
{}
samuel@nb-lubuntu:~$

scoleri, 2021-05-27:
Mine times out or something when there is no charge.  Do negative to negative on the BMS and see if it comes back on.  I have a DALY Smart 250a 12volt and for whatever reason when its not charging for a certain amount of time it turns off???  Before the sun came up this morning i got the same response as soon as a tiny bit of sun came up it turned back on.

I'm on US East Coast (Atlanta Georgia)

scoleri, 2021-05-27:
I'm getting current now....maybe same issue with power:

```
root@battery:~# daly-bms-cli  -d /dev/ttyUSB0 --sinowealth --soc --cell-voltages --temperatures
{
  "total_voltage": 13.377,
  "current": 3.14,
  "soc_percent": 68.0
}
{
  "1": 3.342,
  "2": 3.344,
  "3": 3.344,
  "4": 3.347
}
{
  "external1": 29.65,
  "external2": 27.31,
  "ic1": 29.74,
  "ic2": 0.71
}
```

SamLue, 2021-05-27:
> Mine times out or something when there is no charge. Do negative to negative on the BMS and see if it comes back on. I have a DALY Smart 250a 12volt and for whatever reason when its not charging for [...]

I thought only the Bluetooth Dongle is turned off after some time. But you are right. If I shorten the two PIN's to wake up the BMS I'm getting an answer.
If I remember correctly with the sinowealth-software it is not necessary to wake up the BMS before. Maybe there is a serial message which can wake up the BMS!?

I've checked the values against the Bluetooth-App.
total_voltage, soc_percent and all cell voltages are fine.

But the temperatur is not correct.

samuel@nb-lubuntu:~/daly/python-daly-bms$ daly-bms-cli -d /dev/ttyUSB0 --sinowealth --temperatures
{
  "external1": 29.28,
  "external2": 27.31,
  "ic1": 22.31,
  "ic2": 0.71
}
samuel@nb-lubuntu:~/daly/python-daly-bms$

In the Smartphone-App via Bluetooth the Temperatur is
T1: 19°C

scoleri, 2021-05-27:
Sometimes when i put a large load on the battery this returns:

current?

```
{
  "total_voltage": 14.151,
  "current": 4294953.151,
  "soc_percent": 99
}
{
  "1": 3.58,
  "2": 3.58,
  "3": 3.572,
  "4": 3.424
}

dreadnought, 2021-05-27:
I've pushed and update for the temperature and current values. I guess the current value is negative while discharging and positive while charging, so a signed, not unsigned integer. For the temperature I'm doing -10 now, so that `external1` matches the value that you see in the Bluetooth app. Let's see if that work under all conditions.

I also want to thank you for the heavy testing. Just a small note: If you see wrong values, please run it again with `--verbose`, so that I can see the raw response that comes from the BMS.

Are you seeing any other values in the app that you're interested in?

scoleri, 2021-05-27:
I've updated - still working - its raining now so i'm not charging probably.  

It'd be cool to see the current as + or - (in/out).

Mosfet/balancing would be cool too.

Not sure what your nagios output looked like but that'd be cool too.

scoleri, 2021-05-27:
actually now i'm seeing - (negative) with load
```
{
  "total_voltage": 13.342,
  "current": -12.72,
  "soc_percent": 99
}
{
  "1": 3.341,
  "2": 3.344,
  "3": 3.34,
  "4": 3.32
}

scoleri, 2021-05-27:
Example of load going on and off:
```
timestamp,totalvoltage,current,soc
20210527-180824,14.737,7.405,100
20210527-180829,14.692,,100
20210527-180835,14.667,,100
20210527-180841,14.644,,100
20210527-180847,14.626,,100
20210527-180853,14.617,,100
20210527-180859,14.603,,100
20210527-180904,14.467,-14.669,100
20210527-180910,14.377,-11.445,100
20210527-180916,14.311,-10.62,99
20210527-180922,14.255,-8.979,99
20210527-180928,14.223,-7.93,99

scoleri, 2021-05-28:
Just realized the temps are broken now:

```
root@battery:~# daly-bms-cli -d /dev/ttyUSB0 --sinowealth --temperatures --verbose
Traceback (most recent call last):
  File "/usr/local/bin/daly-bms-cli", line 4, in <module>
    __import__('pkg_resources').run_script('dalybms==0.1.0', 'daly-bms-cli')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.6/dist-packages/dalybms-0.1.0-py3.6.egg/EGG-INFO/scripts/daly-bms-cli", line 73, in <module>
  File "/usr/local/lib/python3.6/dist-packages/dalybms-0.1.0-py3.6.egg/dalybms/daly_sinowealth.py", line 136, in get_temperatures
ValueError: too many values to unpack (expected 2)

dreadnought, 2021-05-30:
> Mosfet/balancing would be cool too.

From the commands that I can see in the Windows software, there are a few left which I haven't implement so far:
```
11 = Full Charge Capacity
12 = Remaining Capacity
14 = Cycle Count
15 = Pack Status
16 = Battery Status
17 = Pack Config
18 = Manufacture Access
```
From the data dump that I got from Sam in the beginning I can tell that 11&12 are just number, but for the other commands I've no idea what the bits and bytes in the returned response mean. 

> Not sure what your nagios output looked like but that'd be cool too.

Have you tried `--check`? It doesn't really check something at the moment, it's basically returning the voltage and current as performance data, so that the monitoring can show them in diagrams. But as written above, I don't know what the status codes of the Sinowealth BMS mean, so adding a real check is hard without documentation.

> Just realized the temps are broken now:

That should be fixed.

scoleri, 2021-06-04:
Yeah temp works now.  --check works great thanks.  Those other things are interesting but i have what i need for now thank you.

SamLue, 2021-06-08:
I have still the wrong value for the temperature.

```
samuel@nb-lubuntu:~/daly/python-daly-bms$ sudo daly-bms-cli  -d /dev/ttyUSB0 --verbose --sinowealth --temperatures
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0c\x02'), 0a0c02
DEBUG    [daly_sinowealth.py:91] 0b7d00 3
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\r\x02'), 0a0d02
DEBUG    [daly_sinowealth.py:91] 0aab2f 3
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0e\x02'), 0a0e02
DEBUG    [daly_sinowealth.py:91] 08b76b 3
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0f\x02'), 0a0f02
DEBUG    [daly_sinowealth.py:91] 00470b 3
{
  "external1": 29.41,
  "external2": 27.31,
  "ic1": 22.31,
  "ic2": 0.71
}

```

I also recognized that there is no current value with param --soc

```
samuel@nb-lubuntu:~/daly/python-daly-bms$ sudo daly-bms-cli  -d /dev/ttyUSB0 --verbose --sinowealth --soc
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0b\x02'), 0a0b02
DEBUG    [daly_sinowealth.py:91] 34063e 3
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x10\x04'), 0a1004
DEBUG    [daly_sinowealth.py:91] 0000000042 5
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x13\x02'), 0a1302
DEBUG    [daly_sinowealth.py:91] 005cc5 3
{
  "total_voltage": 13.318,
  "soc_percent": 92.0
}
```

```
samuel@nb-lubuntu:~/daly/python-daly-bms$ git pull
Bereits aktuell.
samuel@nb-lubuntu:~/daly/python-daly-bms$ git status
Auf Branch sinowealth
Ihr Branch ist auf demselben Stand wie 'origin/sinowealth'.

nichts zu committen, Arbeitsverzeichnis unverändert
```
All code is up-to-date and branch is sinowealth.

What's wrong with my installation?
Yes, I checked local path ~/.local/bin There is no daly-bms-cli anymore.

dreadnought, 2021-06-13:
I've fixed the issue that the `current` value was missing it was zero, e.g. when the BMS is idle, pull the latest changes to get it.

Regarding the temperature, I think that you're still running an outdated version somehow. Run `sudo whereis daly-bms-cli` and it will tell you where you're running it from. If you have installed it with sudo, it should be in `/usr/local/bin/daly-bms-cli`. When you want to update it there, you have to run `python3 setup.py install` with `sudo` too, otherwise it will install it in your user account.

SamLue, 2021-06-13:
OK,

now current with 0.0 is in the result. Thanks for the fix.

{
  "total_voltage": 13.32,
  "current": 0.0,
  "soc_percent": 92.0
}

So, first of all for me the important info that I have the latest Version.

But why are the temperture-values still wrong?

dreadnought, 2021-06-13:
> So, first of all for me the important info that I have the latest Version.
> 
> But why are the temperture-values still wrong?

Yes, you're right about that. Please give it another try now, pull again, install, get the temperatures and compare them to what you see in the app.

SamLue, 2021-06-13:
Mh, sorry!
Still not correct values.

```
samuel@nb-lubuntu:~/daly/python-daly-bms$ sudo daly-bms-cli  -d /dev/ttyUSB0 --verbose --sinowealth --temperatures
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0c\x02'), 0a0c02
DEBUG    [daly_sinowealth.py:91] 0b9596 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\r\x02'), 0a0d02
DEBUG    [daly_sinowealth.py:91] 0aab2f (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0e\x02'), 0a0e02
DEBUG    [daly_sinowealth.py:91] 08b76b (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0f\x02'), 0a0f02
DEBUG    [daly_sinowealth.py:91] 00470b (3)
{
  "external1": 19.65,
  "external2": 17.31,
  "ic1": 12.309999999999999,
  "ic2": -9.29
}
```

In the App 23°C
[image]

dreadnought, 2021-06-16:
Thanks for the details, but it's a bit tricky to fix it.

That's the raw data we have so far:

|  external1  | external2 | ic1 | ic2 | App |
| -------------- | ------------- | --- | --- | --- |
| 2965 | 2731 | 2974 | 71 | ? |
| 2928  | 2731  | 2231  | 71  | 19 °C |
| 2941 | 2731 | 2231 | 71 | ? |
| 2965 | 2731  | 2231 | 71 | 23  °C  |

* The values in all the output that you got so far were shown divided by 100, as it seemed to be the closest match
* The last one had a change applied that subtracted 10 from all temperature values, that's why my table has different values than your commit

So we know that `external2` and `ic2` are useless because they ever change. `external1` is moving, but not linear to the temperature show in the app (2928 -> 19 °C, 2965 -> 23  °C). `ic1` seems to be fix for each BMS, as the first line is from  scoleri and the other three lines are from you.

The BMSTool Windows software could reveal which of the 4 sensors matches the T1 value in the Bluetooth app. Then we still need to figure out how the conversion gets calculated.

SamLue, 2021-06-24:
OK,

I know that my BMS has only one external temp sesor.

So, can you try it with the following calculation.

(external1 - external2)/10
Example: (2965 - 2731)/10 = 234/10 = 23,4 °C

If you have implemented this, I will check some extreme values between App and this calc.

Maybe 2731 is a reference value for 0?

dreadnought, 2021-06-26:
You've found out that the values are in Kelvin (273,1 K = 0 °C), very good. I've applied a change that implements the conversion from Kelvin to °C, please give it a try.

scoleri, 2021-06-26:
Still working ok for me.
`root@battery:~# daly-bms-cli -d /dev/ttyUSB0 --sinowealth --soc --cell-voltages --temperatures
{
  "total_voltage": 13.538,
  "current": 13.695,
  "soc_percent": 97.0
}
{
  "1": 3.373,
  "2": 3.383,
  "3": 3.375,
  "4": 3.407
}
{
  "external1": 25.80000000000001,
  "external2": 0.10000000000002274
}`

closed 2021-06-28 by dreadnought

dreadnought, 2021-06-28:
I've merged the changes to the main branch and released v0.2.0. Feel free to open a new ticket, if you run into any problems.

DocBrown101, 2021-06-28:
Thank you, now it works for me too! I hope the missing values will still be added :)

JohannesF89, 2022-06-27:
Hi,
first of all thanks for the great work and great piece of software.
I own a Daly BMS 40A BT, 8S LiFePo4 24V and connected a raspberry through the serial port

my question is, do somebody know why I get just empty results from the bms?

pi@raspberrypi:~/python-daly-bms $ daly-bms-cli  -d /dev/serial0 --verbose --sinowealth --soc
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x0b\x02'), 0a0b02
DEBUG    [daly_sinowealth.py:115] empty response for command b
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x10\x04'), 0a1004
DEBUG    [daly_sinowealth.py:115] empty response for command 10
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x13\x02'), 0a1302
DEBUG    [daly_sinowealth.py:115] empty response for command 13
{}

or maybe what I'm doing wrong?

JohannesF89, 2022-06-27:
also tried with --uart.

but still no response

PR #2 (2021-06-28): Support for 3S-4S Daly BMS with Sinowealth chip
Fixes #1