dreadnought/python-daly-bms #3: [Enhancement] Turn on and off mosfets

vicgwee, 2021-07-01:
Thanks for this repo! I was looking for ways to interface with my Daly BMS. 

Wondering if you have worked/are working on a feature to turn on and off the discharge/charge mosfets? Thank you!

dreadnought, 2021-07-04:
What I need to implement it are the commands that the Windows software sends to the BMS over the serial connection, while it gets turned off and on in there. Have you used the official Daly BMS Monitor software already and do you know how to control the mosfets with it?

vicgwee, 2021-07-05:
Yup I've sniffed the packets that the software sent and it works to turn on and off the mosfets. Seems like the command is `D9` and byte 5 decides whether the state is on or off.

Turn on discharge mosfet
Sent: `A5 40 D9 08 01 00 00 00 00 00 00 00 C7` 
Received: `A5 01 D9 08 01 0D 13 0D 21 0D 17 C0 BA`

Turn off discharge mosfet
Sent: `A5 40 D9 08 00 00 00 00 00 00 00 00 C6`
Received: `A5 01 D9 08 00 0D 13 0D 21 0D 17 C0 B9`

dreadnought, 2021-07-06:
Thanks for providing the command. Please check out my latest commit: https://github.com/dreadnought/python-daly-bms/commit/cddafd09dbe1115f085d006ff46ad9ee21b97794

This is how it works for me:
```
# daly-bms-cli  -d /dev/ttyUSB0 --mosfet
{
  "mode": "charging",
  "charging_mosfet": true,
  "discharging_mosfet": true,
  "capacity_ah": 5.97
}
# daly-bms-cli  -d /dev/ttyUSB0 --set-discharge-mosfet on
# daly-bms-cli  -d /dev/ttyUSB0 --set-discharge-mosfet off
# daly-bms-cli  -d /dev/ttyUSB0 --mosfet
{
  "mode": "stationary",
  "charging_mosfet": false,
  "discharging_mosfet": false,
  "capacity_ah": 5.7
}
# daly-bms-cli  -d /dev/ttyUSB0 --set-discharge-mosfet on
# daly-bms-cli  -d /dev/ttyUSB0 --mosfet
{
  "mode": "stationary",
  "charging_mosfet": true,
  "discharging_mosfet": true,
  "capacity_ah": 5.7
}
```
My BMS has only one connector, I think that's why it's changing both at the same time.

I highly suggest to try the `on` command first while it's already on. As long as it exits without an error, also the `off` command should work, but if the `on` fails, don't continue. If you notice any issue, run the command again with `--verbose` and provide me the output.

Frederic-GH, 2021-10-02:
Hi, thanks for your nice work!
I've tried to switch the discharge mosfet and it works with Daly 8s 80A BMS (switches only discharge mosfet).
Is there a chance this would also work with the charge mosfet? It is available in the Daly-Android-App and I can confirm that it works there. 

Any chance that one of you @vicgwee or @dreadnought already sniffed the packages for the charge mosfet also? 

I can not work with the windows software out of the box (all linux boxes here), so I would have to get a Windows PC or a VM running for package sniffing. Which tools would I need to sniff the serial (???) communication. 
Regards

Frederic-GH, 2021-10-03:
Sems like the command is DA for charging mosfet instead of D9 for discharging mosfet. I could extract the value from the Daly-Monitor software (c# windows exe)

Bagarre, 2022-01-12:
Hello
Was there a reason the --set_discharge_mosfet option is not available when using --sinowealth?
The feature is available in the Daly bluetooth app as well as the Sinowealth app.
If it was just an over sight, I'd be happy to add the code.
BTW thanks for building this. Planning to use it to extend some automation. Will be happy to share once i have something.

dreadnought, 2022-01-16:
Hi Bagarre,
the two types of BMS use a completely different protocol for the communication with the app, that's why every command has to be captured and implemented twice here. As I'm using just a single BMS, everything that goes beyond the capabilities of my device is tricky to implement and test.

Bagarre, 2022-01-16:
The iphone app is able to enable/disable charge and discharge for the sinowealth model however the Sinowealth Win32 app does not. Do you know of an easy way to sniff bluetooth traffic on an iphone? I could capture the packets and contribute the features.

sand1812, 2022-12-08:
Hi ; Does anyone know protocol command to set SoC percent ?
My BMS drifts sometimes, and I'd like to be able to re-set SoC via command line.
If I succeed, I'll propose a code update for this.

dreadnought, 2023-11-16:
Support for toggling the charge mosfet was added in #45

closed 2023-11-16 by dreadnought