dreadnought/python-daly-bms #4: running in windows?

spilegi, 2021-07-08:
Will this run in linux only?  It's working great on my Raspberry PI's, but not withing windows.

dreadnought, 2021-07-08:
I've not tested it yet, but theoretically it should work. How did you try to execute it? What output did you get?

pySerial is used for the serial connection. After taking a look at their [documentation](https://pyserial.readthedocs.io/en/latest/shortintro.html#opening-serial-ports) I think that using `COM1` (or whatever number your COM port has) instead of `/dev/ttyUSB0` for the `-d` parameter should be work on Windows.

spilegi, 2021-07-09:
@dreadnought I apologize it does work in windows.  I'm not as familiar with python in windows as I am with it in linux.  

For anyone else that come across this, pip defaults it to: C:\Users\[username]\AppData\Roaming\Python\Python39\Scripts which in not in the path file by default  Once I ran it from the directory it worked.  

[image]

@dreadnought thanks for the software.  I had been using another github project sending the data to a MQTT as a necessary step, this will allow be to put it right in the DB

closed 2021-07-09 by spilegi