dreadnought/python-daly-bms #9: Sinowealth does to seem to support battery packs more than 10s

hacksics, 2021-08-15:
It looks like BMSTool - Sinowealth only supports up to monitoring of 10 cells. And when I connected my BMS it shows all incorrect results. For example total voltage seems to be the voltage of 11th cell and SOC and temperature are all wrong. 

I did some changes to sinowealth python script for it to iterate through 16 cells and it shows up to 15 cells and then again 16th cell is wrong

```
$ daly-bms-cli -d /dev/cu.usbserial-1440 --sinowealth --cell-voltages --verbose
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x01\x02'), 0a0102
DEBUG [daly_sinowealth.py:91] 0e7dbf (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 1 is 3709
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x02\x02'), 0a0202
DEBUG [daly_sinowealth.py:91] 0e7d85 (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 2 is 3709
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x03\x02'), 0a0302
DEBUG [daly_sinowealth.py:91] 0e7d93 (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 3 is 3709
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x04\x02'), 0a0402
DEBUG [daly_sinowealth.py:91] 0e7ef8 (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 4 is 3710
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x05\x02'), 0a0502
DEBUG [daly_sinowealth.py:91] 0e7fe9 (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 5 is 3711
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x06\x02'), 0a0602
DEBUG [daly_sinowealth.py:91] 0e7ddd (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 6 is 3709
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x07\x02'), 0a0702
DEBUG [daly_sinowealth.py:91] 0e7ec2 (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 7 is 3710
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x08\x02'), 0a0802
DEBUG [daly_sinowealth.py:91] 0e7d19 (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 8 is 3709
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\t\x02'), 0a0902
DEBUG [daly_sinowealth.py:91] 0e80f2 (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 9 is 3712
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\n\x02'), 0a0a02
DEBUG [daly_sinowealth.py:91] 0e7e3c (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 10 is 3710
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x0b\x02'), 0a0b02
DEBUG [daly_sinowealth.py:91] 0e7e2a (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 11 is 3710
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x0c\x02'), 0a0c02
DEBUG [daly_sinowealth.py:91] 0e80bc (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 12 is 3712
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\r\x02'), 0a0d02
DEBUG [daly_sinowealth.py:91] 0e7f59 (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 13 is 3711
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x0e\x02'), 0a0e02
DEBUG [daly_sinowealth.py:91] 0e7f63 (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 14 is 3711
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x0f\x02'), 0a0f02
DEBUG [daly_sinowealth.py:91] 0e7f75 (3)
DEBUG [daly_sinowealth.py:102] Response data for cell 15 is 3711
DEBUG [daly_sinowealth.py:61] message: bytearray(b'\n\x10\x04'), 0a1004
DEBUG [daly_sinowealth.py:91] 0e7f0000a6 (5)
DEBUG [daly_sinowealth.py:102] Response data for cell 16 is 243204096
```
In this you can see 0B is cell voltage of 11th cell now.

Given that PC application only supports up to 10 cells, I've used android phone to capture bluetooth messages 
[btsnoop_hci.log](https://github.com/dreadnought/python-daly-bms/files/6987567/btsnoop_hci.log). I don't know how to read this but android application shows data correctly.

hacksics, 2021-08-15:
It looks like 17 is the battery pack voltage in mV. I checked the value with mobile app and it is consistent 
```
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x11\x04'), 0a1104
DEBUG    [daly_sinowealth.py:91] 0000e7e9d2 (5)
DEBUG    [daly_sinowealth.py:102] Response data for cell 17 is 59369
```

hacksics, 2021-08-15:
More debugging. Now I'm able to get SOC
```
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x11\x04'), 0a1104
DEBUG    [daly_sinowealth.py:91] 0000e7e8d5 (5)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x12\x04'), 0a1204
DEBUG    [daly_sinowealth.py:91] 0000000010 (5)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x18\x02'), 0a1802
DEBUG    [daly_sinowealth.py:91] 003557 (3)
{
  "total_voltage": 59.368,
  "current": 0.0,
  "soc_percent": 53.0
}
```

hacksics, 2021-08-15:
Here are my findings: 

**For individual cell voltages** 

Byte 1: `0a`
Byte 2: `start with 01 and can be number of cells in the pack`
Byte 3: `02`

If we apply above then I can pull all the cell voltages after modifying `daly_sinowealth.py` 
```
$ python bin/daly-bms-cli -d /dev/cu.usbserial-1440 --sinowealth --cell-voltages --verbose
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x01\x02'), 0a0102
DEBUG    [daly_sinowealth.py:91] 0e7dbf (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x02\x02'), 0a0202
DEBUG    [daly_sinowealth.py:91] 0e7d85 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x03\x02'), 0a0302
DEBUG    [daly_sinowealth.py:91] 0e7e9a (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x04\x02'), 0a0402
DEBUG    [daly_sinowealth.py:91] 0e7ef8 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x05\x02'), 0a0502
DEBUG    [daly_sinowealth.py:91] 0e7fe9 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x06\x02'), 0a0602
DEBUG    [daly_sinowealth.py:91] 0e7ed4 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x07\x02'), 0a0702
DEBUG    [daly_sinowealth.py:91] 0e7fc5 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x08\x02'), 0a0802
DEBUG    [daly_sinowealth.py:91] 0e7d19 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\t\x02'), 0a0902
DEBUG    [daly_sinowealth.py:91] 0e80f2 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\n\x02'), 0a0a02
DEBUG    [daly_sinowealth.py:91] 0e7f3b (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0b\x02'), 0a0b02
DEBUG    [daly_sinowealth.py:91] 0e7f2d (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0c\x02'), 0a0c02
DEBUG    [daly_sinowealth.py:91] 0e80bc (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\r\x02'), 0a0d02
DEBUG    [daly_sinowealth.py:91] 0e7f59 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0e\x02'), 0a0e02
DEBUG    [daly_sinowealth.py:91] 0e7f63 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0f\x02'), 0a0f02
DEBUG    [daly_sinowealth.py:91] 0e7f75 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x10\x02'), 0a1002
DEBUG    [daly_sinowealth.py:91] 0e7fc0 (3)
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x11\x02'), 0a1102
DEBUG    [daly_sinowealth.py:91] 00007a (3)
{
  "1": 3.709,
  "2": 3.709,
  "3": 3.71,
  "4": 3.71,
  "5": 3.711,
  "6": 3.71,
  "7": 3.711,
  "8": 3.709,
  "9": 3.712,
  "10": 3.711,
  "11": 3.711,
  "12": 3.712,
  "13": 3.711,
  "14": 3.711,
  "15": 3.711,
  "16": 3.711
}
```

dreadnought, 2021-08-20:
Thanks for reporting this issue. I wasn't aware that Daly sells BMS for so many cells, I just thought it was max. 10 because of the Windows app.

Did solve it by increasing `max_cells` to 16 or did find a way to extract the number of cells from somewhere else? If we don't find a way to do it dynamically, I could add a parameter that replaces the hardcoded value and adds an offset to all other commands.

Regarding the btsnoop_hci.log, you can open this type of file with Wireshark to see the communication in a nice GUI. But I don't see data in there that look familiar to the protocol that is used on the wired connection.

hacksics, 2021-08-21:
Somehow I was able to fiddle with BMS Tool and get it to show 16s. Here is the attached serial monitor file 
[daly16s.txt](https://github.com/dreadnought/python-daly-bms/files/7025613/daly16s.txt)

birdie1, 2021-08-29:
Daly is selling BMS upto 24S on aliexpress right now. If you are increasing max_cells, probably increase it to 24.

Carstijn, 2021-09-06:
There are two lines of product, at least in the protocol:
SH367303 / 367305 / 367306 / 39F003 / 39F004 / BMS_10 chips with up to 10 cells
Where the number of cells can be read from the packconfigmap, no room for up to 10 cells
[image]

SH367308 / 367309 / 39F005 / BMS_16 chips with up to 16 cells?
For these chips there is a different definition of the packconfigmap, without a number of cells in the data.

Over the past weeks, I have done some homework and implemented the sinowealth protocol in to the dbus-serialbattery code, see this pull request. https://github.com/Louisvdw/dbus-serialbattery/pull/41

As far as I can see, my code supports reading more data than the current code here.