dreadnought/python-daly-bms #10: Is it possible to use our cmd tool to set certain parameters?

liufuyang, 2021-08-20:
Firstly thanks for the great work, it works out of the box for me.

My BMS arrived without a bluetooth module, so I cannot connect it with my phone to setup it's parameters and that is why the SOC reading seems wrong.

I am not sure exactly how to correct it yet but it would be nice if we could support setting some important values as the bluetooth app (or the pc software) can do on it.

I also don't have a windows pc so cannot use the pc tool they provided...

dreadnought, 2021-08-20:
I would assume that the SOC gets calibrated automatically after a few charge/discharge cycles, but I don't know for sure. My tool doesn't have anything implemented to set parameters of the BMS, mainly because there is no documentation for it. But it might be possible by capturing the commands that the Windows software is sending and trying to understand their structure. 

Did you have to set `--sinowealth` to make this tool work or do you have a "normal" Daly BMS?

Here is a screenshot of the parameter settings page of the DalyBMS Windows app:
[image]

Which parameter would suit your needs?

liufuyang, 2021-08-20:
Thank you. 

About which parameter I need to set: I am not exactly sure as I haven't played with it before and there seems no documents or tutorials about it. But what I feel at least if we could set those things then it could be very helpful (in the case where there is no bluetooth connected)
(or at least it would be nice if we could read those settings out first)
* rated cap (Ah) --- which might help make SOC more precise.
* rated cell v --- which could be nice to double check whether this setting is correct
*  balancer start v - which is nice to know
*  balancer start diff v - which nice to know
*  cell volt high/low - which is nice to know/set
*  sum volt high/low - which is nice to know and set, so I could set to stop charging at certain voltage and stop discharging with it reach certain voltage, then I could make sure the battery works between 20% to 80% SOC all the time.

I know this perhaps sounds like asking a lot, but I guess perhaps we can try start showing only some of those values. I guess they are not read for the current version? 

I have a MGod 10S LTO BMS, and MGod is basically Daly but branded with another name.  Sorry I was wrong about with or without --sinowealth it both works for me. Testing it again apparently it only works for me without it :)

<details><summary>Come command line output on my side:</summary>
<p>

```
(master #%) 🦀 👉  daly-bms-cli -d /dev/tty.usbserial-110 --all
{
  "soc": {
    "total_voltage": 23.2,
    "current": -2.3,
    "soc_percent": 64.0
  },
  "cell_voltage_range": {
    "highest_voltage": 2.348,
    "highest_cell": 3,
    "lowest_voltage": 2.294,
    "lowest_cell": 7
  },
  "temperature_range": {
    "highest_temperature": 24,
    "highest_sensor": 1,
    "lowest_temperature": 24,
    "lowest_sensor": 1
  },
  "mosfet_status": {
    "mode": "discharging",
    "charging_mosfet": true,
    "discharging_mosfet": true,
    "capacity_ah": 19.2
  },
  "status": {
    "cells": 10,
    "temperature_sensors": 1,
    "charger_running": false,
    "load_running": false,
    "states": {
      "DI1": false,
      "DI2": true
    },
    "cycles": 1
  },
  "cell_voltages": {
    "1": 2.315,
    "2": 2.336,
    "3": 2.348,
    "4": 2.318,
    "5": 2.314,
    "6": 2.324,
    "7": 2.294,
    "8": 2.318,
    "9": 2.329,
    "10": 2.313
  },
  "temperatures": {
    "1": 24
  },
  "balancing_status": {
    "error": "not implemented"
  },
  "errors": []
}

```

</p>
</details>

liufuyang, 2021-08-25:
I finally found a Windows machine (via a virtual machine) and it did the job for me... So I don't really need this feature for now, but I think it would be nice if we could read and set some important parameters and it shouldn't be that difficult I guess.

BTW I made [a YouTube video](https://www.youtube.com/watch?v=k20iQ_k224E&t=344s) to record how I did it, if anyone else in interested in this matter (only have an intel mac and no windows machine around you for example)

sand1812, 2022-12-21:
Hi ; I just added set_soc to the lib (see pull_request).
It would be easy to add it to command line interface.