dreadnought/python-daly-bms #11: Precision

n-benchoubane, 2021-09-09:
Hi!

I was reading this post  [here ](https://diysolarforum.com/threads/daly-bms-in-stock-in-usa.20701/)

It is mentioned that they calibrate the voltage and current measurements with a **6 1/2 digit resolution** HP/Agilent meter but from the program, this precision isn't displayed. Is there a way to modify and get the higher precision?

Thank you always!
Nesrine

dreadnought, 2021-09-11:
Hi,
the program doesn't round up the values, it displays them with the same precision that gets from the BMS and as the Windows app does. 

For which use-case do you need a higher precision? I'm using the data from the BMS to set limits for other devices (inverter, charger), but my battery pack is so big that I don't rely on multi-digit decimal values.

n-benchoubane, 2021-09-15:
Hello!

I did a more thorough analysis and was able to confirm that it is due to the chip itself.
Thanks for the quick response!

closed 2021-09-15 by n-benchoubane