dreadnought/python-daly-bms #16: Error due to changes 3~4 days ago

piano22, 2021-09-23:
It seems like the recent changes in code (adding Homassistant?) produced a syntax error.:

traceback (most recent call last):
  File "/usr/local/bin/daly-bms-cli", line 4, in <module>
    __import__('pkg_resources').run_script('dalybms==0.2.0', 'daly-bms-cli')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 739, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1500, in run_script
    script_code = compile(script_text, script_filename, 'exec')
  File "/usr/local/lib/python3.5/dist-packages/dalybms-0.2.0-py3.5.egg/EGG-INFO/scripts/daly-bms-cli", line 93
    hass_config_topic = f'homeassistant/sensor/daly_bms/{base.replace("/", "_")}/config'
                                                                                       ^
SyntaxError: invalid syntax

dreadnought, 2021-09-23:
It seems that you're using the quite old and end-of-life Python version 3.5. The code line that causes the error uses a f-string, a feature that was introduced in Python 3.6.

Which Linux distribution and version are you using?

piano22, 2021-09-23:
Oh, I'm sorry that I wasted your time. I have a quite old raspi 2 - my first one. And the active python version was actually 2.7;-) The latest version, I can switch to, is 3.5. Looks like I have to upgrade stretch to buster - if it is possible :-(

piano22, 2021-09-24:
As a conclusion: I successfully upgraded Python and PIP to 3.6 (by compiling from source)! And now your tool works like a charm. As a first step I am piping the output with some values of interest through a little perl-script (this is more my world;-) and write comma-separated lines all 5 minutes to a file.
Thank you for your great work!

closed 2021-09-24 by piano22