dreadnought/python-daly-bms #18: NO ISSUE Newbee is looking for a helping / example script

JStefanRaspi, 2021-12-07:
Hello World, sorry but I am a beginner in this thing... would anyone give me a helping hand and show me how to use this Dalybms script... I get it to work in the Terminal by -cli.... but how do I get values inside a python script to send it to Blynk or wherever... 
Thank you very much... an example would be more than welcome.
Cheers
Stefan

JStefanRaspi, 2021-12-10:
I am still trying to get that to work... failed up to know. I try to convert the json data into something else... but that's not the way I know.. but I have no glue how to handle the dalybms module and have no idea about and google is not my friend... :-( 
Would be great if someone let me know. THX

icenov, 2021-12-27:
I just got this to work with some minor(?) adjustements (using linux ubuntu...). I use a python script to run daly-bms-cli.py as a subprocess, call in everything (i.e. --all) then parse the returned value back into a dict for further processing.
the key line to use in the new script is (import subprocess):
```
returned_bms =  subprocess.check_output('/usr/bin/python3 /home/tony/projects/daly/daly-bms-cli.py -d /dev/ttyUSB2 --all'.split(), text=True).replace('\n',"")
```
This basically runs the standard daly-bms-cli.py and returns the output as a string (stripping \n). You can then turn it back to a dict with eg bms_all = json.loads(returned_bms)
Make sure all scripts are in the same directory or use absolute paths.
My biggest problem was getting the right USB port and having permission to use it!

Once you get the returned dictionary, it's easy to extract what you need.

JStefanRaspi, 2021-12-28:
Hey Tony, thanks for sharing.. I could not try at the moment. ASAP I let you know .

Happy New Years 
Stefan

Wer, Wo, Womit..... www.tvware.de

JStefanRaspi, 2022-01-05:
Hey... Yes that line 
**_returned_bms = subprocess.check_output('daly-bms-cli -d /dev/ttyUSB0 --all'.split(), text=True)_**
works for some time... but from time to time I get a fault and it stops working... but hopefully I find the mistake... better sooner than later... the error 
**_Subprocess check_output returned non-zero exit status 1_**
could come from that I do not know the full path of the daly-bms-cli... and I have no clue how to find out... what to ask google... :-( 
THX
Stefan

icenov, 2022-01-05:
I have had this happen also and found it was due to the Daly BMS turning off after a time. I made mine stay on indefinitely (it's a large battery so the loss is negligible). It has now been running and sending data for a few days without fault. I used the bluetooth app to change the setting, but it can probably be done with the daly-bms-cli. Dreadnaught has shown how to use the script without subprocess, which might be simpler. See [this issue ](https://github.com/dreadnought/python-daly-bms/issues/20#issuecomment-1003531149)

JStefanRaspi, 2022-01-06:
**/usr/local/bin/daly-bms-cli** is the path of the cli file

closed 2022-01-11 by JStefanRaspi