dreadnought/python-daly-bms #20: ResolutionError: Script 'scripts/daly-bms-cli' not found in metadata

icenov, 2021-12-22:
Hi
I'm using python 3.9 in Ubuntu 20.04 and trying to connect to a Daly BMS using USB, but getting this error (using sudo but same error as user):
tony@tony-acer:~$ sudo daly-bms-cli  -d /dev/ttyUSB0 --sinowealth
[sudo] password for tony: 
Traceback (most recent call last):
  File "/usr/local/bin/daly-bms-cli", line 4, in <module>
    __import__('pkg_resources').run_script('dalybms==0.3.0', 'daly-bms-cli')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 651, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1436, in run_script
    raise ResolutionError(
pkg_resources.ResolutionError: Script 'scripts/daly-bms-cli' not found in metadata at '/usr/local/lib/python3.9/dist-packages/dalybms-0.3.0.dist-info'

I installed using pip3, but it seems a script is missing? Can you suggest a solution?
Thanks

dreadnought, 2021-12-22:
I've tried to install it both ways for Python 3.9, via pip and manually, but for me it works fine.

That's what I did on a fresh Ubuntu 20.04 system:
```
apt install python3.9 python3-pip
# via pip
python3.9 -m pip install dalybms
# or manually
git clone git@github.com:dreadnought/python-daly-bms.git
cd python-daly-bms/
python3.9 -m pip install pyserial
python3.9 setup.py install
```

You wrote that you've installed it with pip3, so maybe it's only installed for Python 3.8, the default version on Ubuntu 20.04. Try to install it explicitly for Python 3.9, as you can see above.

icenov, 2021-12-22:
OK - I did try as per above, but all seems to be installed already, so doesn't seem to a python version issue:

tony@tony-acer:~$ sudo apt install python3.9 python3-pip
[sudo] password for tony: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3.9 is already the newest version (3.9.7-2build1).
python3.9 set to manually installed.
python3-pip is already the newest version (20.3.4-4).
0 to upgrade, 0 to newly install, 0 to remove and 10 not to upgrade.
tony@tony-acer:~$ python3.9 -m pip install dalybms
Requirement already satisfied: dalybms in ./.local/lib/python3.9/site-packages (0.3.0)
tony@tony-acer:~$ python3.9 -m pip install pyserial
Requirement already satisfied: pyserial in ./.local/lib/python3.9/site-packages (3.5)

I will reinstall using a new environment and see if that makes a difference.
Thanks

dreadnought, 2021-12-23:
I think I see the difference already. In your first message the command was run as root (with sudo) and the output was saying that it can't find it in `/usr/local/lib/python3.9/dist-packages`. But your last message shows that it was installed in the users home directory in `~/.local/lib/python3.9/site-packages`. So when root looks for it, it doesn't find it.

icenov, 2021-12-25:
Thanks - OK, I see the error now. I was running as root because I couldn't access /dev/ttyUSB0 as user. If I run the command as user I need to change usb port permission to 600 and it runs well, but doesn't persist. I'm looking at how to permanently change this atm. The other problem now is if I setup a cron job to run every minute, it fails because it also can't find the missing package - presumably because cron is not running as me (i.e.user).

icenov, 2021-12-27:
Sorted now - I gave up on using cron for the moment and just using a while loop in python. As long as I change permission on the usb port it works fine. I run another small script that uses python subprocess to call in the daly-bms-cli script with --all, then parses the return value back to a dictionary that gets imported into a DataFrame, finally saved as a csv file. Thanks so much for your script!

dreadnought, 2021-12-30:
Great that you've figured out a solution that works for you, I just want to comment on two things:

1. There are two easy ways to run a cronjob as a non-root user, you either edit the users crontab with `crontab -e` while you're logged that user, or by putting a file `/etc/cron.d/` that looks like this:
```
# Example of job definition:
# .---------------- minute (0 - 59)
# |  .------------- hour (0 - 23)
# |  |  .---------- day of month (1 - 31)
# |  |  |  .------- month (1 - 12) OR jan,feb,mar,apr ...
# |  |  |  |  .---- day of week (0 - 6) (Sunday=0 or 7) OR sun,mon,tue,wed,thu,fri,sat
# |  |  |  |  |
# *  *  *  *  * user-name command to be executed
17 *	* * *	tony    cd /home/tony/ && daly-bms-cli -d /dev/ttyUSB0 --sinowealth ...
```
After you've done that, have a look at the syslog to see if the cron process reports any syntax errors.

2. You don't need to run it a subprocess when your script is written in Python too, you basically need to take 5 lines from cli script:
```
import logging
from dalybms import DalyBMSSinowealth
logger = logging.getLogger()
bms = DalyBMSSinowealth(request_retries=args.retry, logger=logger)
result = bms.get_all()
```
`result` would then be the same Dict you would get if you run json.loads() with the output of the subprocess.

icenov, 2021-12-31:
Thanks - I did try the non-root crontab approach as suggested but didn't have 'cd'! I think that must have been the problem. Yes, just learning about subprocess. I'll give the 5 line script a try with mqtt in a day or 2.

icenov, 2022-01-01:
Won't this need the parser = argparse.ArgumentParser() module/routine as well (request_retries=args.retry,)?
[I'm not using the Sinowealth version]

dreadnought, 2022-01-01:
You're right, I forgot to modify this line. But it's up to you what you configure using command line arguments, read from a config file or hardcode in the code.

Here is an updated minimal example that works for non-Sinowealth BMS:
```
import logging
from dalybms import DalyBMS
logger = logging.getLogger()
bms = DalyBMS(request_retries=5, logger=logger)
bms.connect(device="/dev/ttyUSB0")
result = bms.get_all()
print(result)
```

JStefanRaspi, 2022-01-06:
Hello Dreadnought, first of all thx for your work... 

I tried your code above... and I ran into a import error:
> from .error_codes import ERROR_CODES
Import Error: attemted relative import with no known parent package <

any idea what went wrong or how to fix?

RPI4 -Python 3.7.3- (I tried before with daly-bms-cli & subprocess and that worked for some hours and than hang... following the suggestion from Tony [](https://github.com/dreadnought/python-daly-bms/issues/18)

Thank you Mrs/Sir ?

dreadnought, 2022-01-06:
Hi,
is your small script maybe placed directly in the python-daly-bms directory? If that's the case please try to move it somewhere else. 

If that doesn't help, can you please post the full stacktrace? There should be some more lines around the two lines that you've posted, which might provide useful insides.

JStefanRaspi, 2022-01-06:
Hi, thx for the fas replay
 The script runs outside (in a different foulder)

The whole error:

> #Traceback (most recent call last):
>   #File "/home/pi/Progs/daly_bms_csv3.py", line 5, in <module>
>     #from dalybms import DalyBMS
>   #File "/home/pi/Progs/dalybms.py", line 7, in <module>
>     #from .error_codes import ERROR_CODES
> #ImportError: attempted relative import with no known parent package

THX Stefan

dreadnought, 2022-01-06:
It looks like you've taken the `dalybms.py` out of the modules directory, I think that's breaking it. The easiest way to fix it should be to install the module as it's described in the [README](https://github.com/dreadnought/python-daly-bms) and remove your copy of the `dalybms.py` that is lying next to your `daly_bms_csv3.py`. If installing it is not an option for you, have at least a directory called `dalybms`, where you put the `dalybms.py` and `error_codes.py`.

JStefanRaspi, 2022-01-06:
pi@Raspi104:~ $ pip3 install dalybms
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: dalybms in ./.local/lib/python3.7/site-packages (0.2.0)

The dalybms folder Location: **/home/pi/.local/lib/python3.7/site-packages** 
inside are: **foulder __pycache__ , daly_bms.py, daly_bms_bluetooth.py, daly_sinowealth.py, error_codes.py, __init__.py**

JStefanRaspi, 2022-01-06:
& inside the site-packages folder is: dalybms and dalybms-0.2.0.dist-info

dreadnought, 2022-01-06:
You have way to many copies of it. 

- Delete /home/pi/Progs/Daly/daly_bms.py and /home/pi/Progs/Daly/error_codes.py
- Delete /home/pi/.local/lib/python3.7/site-packages/dalybms (it's the outdated v0.2, installed for the user pi)
- Keep /usr/local/lib/python3.7/dist-packages/dalybms (it's the latest v0.3, installed system wide)
- Run `sudo python3 /home/pi/Progs/daly_bms_csv3.py` and it should automagically find the v0.3.

JStefanRaspi, 2022-01-06:
No.. does not find it. in the progs there is no **daly_bms_csv3** could it be that the dalybms-0.3.0-py3.7.egg is not "unwrapped" still egg??

JStefanRaspi, 2022-01-11:
I want to thank Mr. dreadnought for his work and his extra ordinary Service. 

Thx Man take good care
Stefan

closed 2023-11-16 by dreadnought