dreadnought/python-daly-bms #21: After Update it's not running anymore

Achilles160716, 2021-12-26:
I don't know why i get this:

~ $ daly-bms-cli -d /dev/ttyUSB0 --sinowealth --all
Traceback (most recent call last):
  File "/usr/local/bin/daly-bms-cli", line 183, in <module>
    result = bms.get_all()
  File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 228, in get_all
    "mosfet_status": self.get_mosfet_status(),
  File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 198, in get_mosfet_status
    if pack_response[key] == "1":
TypeError: 'bool' object is not subscriptable

BR
Achilles

dreadnought, 2021-12-26:
Hi,
please let me know what exactly you've updated and which BMS model you're using. 

Additionally it would be helpful if your run the command in verbose mode and post the output here: 
`daly-bms-cli -d /dev/ttyUSB0 --sinowealth --mosfet --verbose`

Achilles160716, 2021-12-26:
Good evening

I have the daly smart BMS 4S 12V 250Ah. 

~ $ daly-bms-cli -d /dev/ttyUSB0 --sinowealth --mosfet --verbose
DEBUG    [daly_sinowealth.py:89] message: bytearray(b'\n\x11\x04'), 0a1104
DEBUG    [daly_sinowealth.py:111] empty response for command 11
DEBUG    [daly_sinowealth.py:89] message: bytearray(b'\n\x12\x04'), 0a1204
DEBUG    [daly_sinowealth.py:111] empty response for command 12
DEBUG    [daly_sinowealth.py:89] message: bytearray(b'\n\x15\x02'), 0a1502
DEBUG    [daly_sinowealth.py:111] empty response for command 15
Traceback (most recent call last):
  File "/usr/local/bin/daly-bms-cli", line 166, in <module>
    result = bms.get_mosfet_status()
  File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 198, in get_mosfet_status
    if pack_response[key] == "1":
TypeError: 'bool' object is not subscriptable

Achilles160716, 2021-12-26:
and this:
daly-bms-cli -d /dev/ttyUSB0 --sinowealth --soc --verbose
DEBUG    [daly_sinowealth.py:89] message: bytearray(b'\n\x0b\x02'), 0a0b02
DEBUG    [daly_sinowealth.py:111] empty response for command b
DEBUG    [daly_sinowealth.py:89] message: bytearray(b'\n\x10\x04'), 0a1004
DEBUG    [daly_sinowealth.py:111] empty response for command 10
DEBUG    [daly_sinowealth.py:89] message: bytearray(b'\n\x13\x02'), 0a1302
DEBUG    [daly_sinowealth.py:111] empty response for command 13
{}

Sorry, how can I go back to the version before?

dreadnought, 2021-12-26:
It seems like none of the commands get answered by the BMS. Is `/dev/ttyUSB0` still the right device or did it change for some reason?

You have to downgrade it by using the same method that you've used for the installation. If you've installed it from Github, go to the directory where the code is, run `git checkout v0.2.0` and call `sudo python3 setup.py install` again. If you've installed it via pip, run `pip3 install dalybms==0.2.0`

Achilles160716, 2021-12-26:
Version 0.2.0 runs, but there was a version where this was fixed:
~ $ daly-bms-cli -d /dev/ttyUSB1 --sinowealth --all
Traceback (most recent call last):
  File "/home/pi/.local/bin/daly-bms-cli", line 82, in <module>
    result = bms.get_all()
AttributeError: 'DalyBMSSinowealth' object has no attribute 'get_all'

Achilles160716, 2021-12-26:
Sorry writing again. I mad some tests and 0.3.0 runs after booting. here is the log in iobroker. I just call it and catch the returns:

22:01:58.411 | info | javascript.0 (1345) Start javascript script.js.common.Test_BMS
22:01:58.450 | info | javascript.0 (1345) script.js.common.Test_BMS: registered 0 subscriptions and 0 schedules
22:01:59.472 | info | javascript.0 (1345) script.js.common.Test_BMS: Result BMS: **{ "current": 0.0, "soc_percent": 64.0 }**
22:01:59.473 | info | javascript.0 (1345) script.js.common.Test_BMS: Error BMS: null
22:02:15.432 | info | javascript.0 (1345) Stop script script.js.common.Test_BMS

22:02:15.444 | info | javascript.0 (1345) Start javascript script.js.common.Test_BMS
22:02:15.488 | info | javascript.0 (1345) script.js.common.Test_BMS: registered 0 subscriptions and 0 schedules
22:02:16.532 | info | javascript.0 (1345) script.js.common.Test_BMS: Result BMS:
22:02:16.532 | info | javascript.0 (1345) script.js.common.Test_BMS: Error BMS: Error: Command failed: daly-bms-cli -d /dev/ttyUSB1 --sinowealth --soc Traceback (most recent call last): File "/usr/local/bin/daly-bms-cli", line 163, in <module> result = bms.get_soc() File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 155, in get_soc return self._read_bulk(requests) File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 142, in _read_bulk response_data = self._read(command[0]) File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 120, in _read return struct.unpack('>h x', response_data)[0] struct.error: unpack requires a buffer of 3 bytes
22:03:10.795 | info | javascript.0 (1345) Stop script script.js.common.Test_BMS

22:03:10.806 | info | javascript.0 (1345) Start javascript script.js.common.Test_BMS
22:03:10.848 | info | javascript.0 (1345) script.js.common.Test_BMS: registered 0 subscriptions and 0 schedules
22:03:11.431 | info | javascript.0 (1345) script.js.common.Test_BMS: Result BMS:
22:03:11.431 | info | javascript.0 (1345) script.js.common.Test_BMS: Error BMS: Error: Command failed: daly-bms-cli -d /dev/ttyUSB1 --sinowealth --soc Traceback (most recent call last): File "/usr/local/bin/daly-bms-cli", line 163, in <module> result = bms.get_soc() File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 155, in get_soc return self._read_bulk(requests) File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 142, in _read_bulk response_data = self._read(command[0]) File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 109, in _read response_data = self.serial.read(length + 1) File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 596, in read 'device reports readiness to read but returned no data ' serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
22:04:07.371 | info | javascript.0 (1345) Stop script script.js.common.Test_BMS

22:04:07.380 | info | javascript.0 (1345) Start javascript script.js.common.Test_BMS
22:04:07.422 | info | javascript.0 (1345) script.js.common.Test_BMS: registered 0 subscriptions and 0 schedules
22:04:07.967 | info | javascript.0 (1345) script.js.common.Test_BMS: Result BMS:
22:04:07.967 | info | javascript.0 (1345) script.js.common.Test_BMS: Error BMS: Error: Command failed: daly-bms-cli -d /dev/ttyUSB1 --sinowealth --soc Traceback (most recent call last): File "/usr/local/bin/daly-bms-cli", line 163, in <module> result = bms.get_soc() File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 155, in get_soc return self._read_bulk(requests) File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 142, in _read_bulk response_data = self._read(command[0]) File "/usr/local/lib/python3.7/dist-packages/dalybms/daly_sinowealth.py", line 109, in _read response_data = self.serial.read(length + 1) File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 596, in read 'device reports readiness to read but returned no data ' serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

Achilles160716, 2021-12-28:
I solved it over simlink

thank you for your perfect support

closed 2021-12-28 by Achilles160716