dreadnought/python-daly-bms #22: No values are displayed

elflinko, 2022-01-08:
After installation on a Raspberry 3B, the first queries worked as expected. Now I get an empty json format without values. What else can I try? Why did it work for a short time?
Output example:
**root@solaranzeige:/home/pi# daly-bms-cli -d /dev/ttyUSB0 --soc**
{
  "total_voltage": 0.0,
  "current": -3000.0,
  "soc_percent": 0.0
}
#+++++++++++++++
**root@solaranzeige:/home/pi# daly-bms-cli -d /dev/ttyUSB0 --soc --verbose        DEBUG    [daly_bms.py:95] -- 94** ------------------------
DEBUG    [daly_bms.py:65] w a5409408000000000000000081
DEBUG    [daly_bms.py:114] 0 a5409408000000000000000081 13
DEBUG    [daly_bms.py:95] -- 90 ------------------------
DEBUG    [daly_bms.py:65] w a540900800000000000000007d
DEBUG    [daly_bms.py:114] 0 a540900800000000000000007d 13
{
  "total_voltage": 0.0,
  "current": -3000.0,
  "soc_percent": 0.0
}

dreadnought, 2022-01-08:
Is maybe another process running that's also connected to the serial device? In that case one of them would get all responses from the BMS, no matter which one send the request.

elflinko, 2022-01-08:
I have reinstalled a Raspberry with an image and nothing else is running. I have installed dalybms on the new OS and pyserial. The only strange thing is that it worked a few times at the beginning and since then not.

elflinko, 2022-01-08:
Very interesting! I restarted the BMS via the mobile app -and it works as usual. THANK YOU for the support!

sudo daly-bms-cli -d /dev/ttyUSB0 --all
{
  "soc": {
    "total_voltage": 26.2,
    "current": 0.0,
    "soc_percent": 90.0
  },
  "cell_voltage_range": {
    "highest_voltage": 3.282,
    "highest_cell": 7,
    "lowest_voltage": 3.281,
    "lowest_cell": 1
  },
  "temperature_range": {
    "highest_temperature": 8,
    "highest_sensor": 1,
    "lowest_temperature": 8,
    "lowest_sensor": 1
  },
  "mosfet_status": {
    "mode": "stationary",
    "charging_mosfet": true,
    "discharging_mosfet": true,
    "capacity_ah": 81.0
  },
  "status": {
    "cells": 8,
    "temperature_sensors": 1,
    "charger_running": false,
    "load_running": false,
    "states": {
      "DI1": false
    },
    "cycles": 0
  },
  "cell_voltages": {
    "1": 3.281,
    "2": 3.282,
    "3": 3.281,
    "4": 3.282,
    "5": 3.281,
    "6": 3.282,
    "7": 3.282,
    "8": 3.281
  },
  "temperatures": {
    "1": 8
  },
  "balancing_status": {
    "error": "not implemented"
  },
  "errors": []
}

closed 2022-01-08 by elflinko