dreadnought/python-daly-bms #23: COM port lock under windows - alternating

icenov, 2022-01-22:
I'm now trying another daly-BMS, which is a 5s -12 V 40A unit and I have managed to get the script to run under windows 10 using ipython, simply running "daly-bms-cli -d COM3 --all". It outputs everything expected, but interestingly it only works every second run. That is if I run the script once and get the results, if I run it again -- say 30 sec later - I get an error message that:
SerialException could not open port 'COM3' access is denied.
But if I rerun the script it now runs. This cycle seems consistent - run once, fail, run once fail etc. It seems that the COM port gets locked and it takes another run to unlock it. It doesn't seem to make any difference how long I wait between runs.
It's not a serious problem as I'm just using windows while I have the batteryy pack on my desk - I'll go back to linux once the power supply is finished.

dreadnought, 2022-01-23:
Hi,
I didn't have the chance to test it on Windows yet, but I've added a small change that might solve your issue. Please get the latest code from Github and try it again.

icenov, 2022-01-23:
Hi
I'm not familiar with windows and couldn't install the new scripts using python3 setup.py install - kept getting permission errors even though I was running as administrator.
So I manually copied all the new files over the top of the existing ones. But now a new error - even though the script runs every second time as before - AttributeError: 'DalyBMS' object has no attribute 'disconnect':
```
In [105]: run daly-bms-cli  -d COM3 --all
{
  "soc": {
    "total_voltage": 12.1,
    "current": 0.0,
    "soc_percent": 99.0
  },
  "cell_voltage_range": {
    "highest_voltage": 2.475,
    "highest_cell": 3,
    "lowest_voltage": 2.411,
    "lowest_cell": 1
  },
  "temperature_range": {
    "highest_temperature": 22,
    "highest_sensor": 1,
    "lowest_temperature": 22,
    "lowest_sensor": 1
  },
  "mosfet_status": {
    "mode": "stationary",
    "charging_mosfet": true,
    "discharging_mosfet": true,
    "capacity_ah": 39.6
  },
  "status": {
    "cells": 5,
    "temperature_sensors": 1,
    "charger_running": false,
    "load_running": false,
    "states": {
      "DI1": false
    },
    "cycles": 0
  },
  "cell_voltages": {
    "1": 2.411,
    "2": 2.463,
    "3": 2.475,
    "4": 2.418,
    "5": 2.424
  },
  "temperatures": {
    "1": 22
  },
  "balancing_status": {
    "error": "not implemented"
  },
  "errors": []
}
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File ~\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\daly-bms-cli:221, in <module>
    218 if mqtt_client:
    219     mqtt_client.disconnect()
--> 221 bms.disconnect()
    223 if not result:
    224     sys.exit(1)

AttributeError: 'DalyBMS' object has no attribute 'disconnect'
```

dreadnought, 2022-01-24:
Copying the `daly-bms-cli` file is not enough, you need to install the new version the same way as the old one. Currently the installed module is still older than the CLI tool, so they are not compatible. Investigate why you get permission errors when running the setup.py.

icenov, 2022-01-25:
Thank you - I did a reinstall using pip (after flushing the cache), but it still has the same issue when using ipython. However when using python3 it runs properly everytime. I suspect it is my installation as I am not a windows user and had a great deal of trouble installing python and ipython. (trouble with powershell etc). 
But for now I will use it as it was supposed to be used! Many thanks.

closed 2023-11-16 by dreadnought