dreadnought/python-daly-bms #25: bluetooth support for sinowealth devices

etienne1911, 2022-04-02:
Hi,

I could successfully serial connect to my BMS which is obviously `sinowealth` as it is only working with this option.

I'm trying to do same over bluetooth following the [given example](https://github.com/dreadnought/energy-storage-controller/blob/master/smart_bms.py).

After some customization, could make the code launch which seems to connect to device, but is not able to request anything then. That would make sense if it expects to use sinowealth protocol..

So I wondered if there was currently any support for bluetooth and sinowealth devices?
Looking at the code, it doesn't seem to be the case. Didn't find a specific class for this purpose nor any options we can provide to switch to it.

Any plan to support it in future?

Unrelated: while connecting in serial mode, current value is always 0. Is it a bug?

dreadnought, 2022-04-04:
Hi,
allowing connections to Sinoweealth devices via Bluetooth was not implemented yet. In case that they use the same communication protocol via Bluetooth than via serial connections, the work that needs do be done is to create a class that overwrites all function calls of the serial connection with the Bluetooth equivalent. 

Here you can see how it works for the "normal" Daly BMS:
https://github.com/dreadnought/python-daly-bms/blob/main/dalybms/daly_bms_bluetooth.py

It takes `DalyBMS` as a parent class and adds the Bluetooth connection, while preserving the public function names. If you want to play around with it, you can replace the references to `DalyBMS` in that file with `DalyBMSSinowealth`, to see if brings you forward. 

I don't have such a BMS and don't have plans to implement the Bluetooth connection myself.

etienne1911, 2022-04-13:
ok thanks for precisions. will eventually have a look when I have a chance
I was thinking there would be bluetooth support for  sinowealth protocol  as there was already serial support