dreadnought/python-daly-bms #31: Characteristic 17 not found

twhtanghk, 2022-10-18:
Run with following error. It seems that characteristics specified is different in my bluetooth sensor. But it reads data correctly on Android app.
```
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/tmp/a.py", line 49, in main
    await con.connect()
  File "/tmp/a.py", line 11, in connect
    await self.bt_bms.connect(mac_address=self.mac_address)
  File "/usr/local/lib/python3.10/dist-packages/dalybms/daly_bms_bluetooth.py", line 45, in connect
    await self.client.start_notify(17, self._notification_callback)
  File "/usr/local/lib/python3.10/dist-packages/bleak/__init__.py", line 639, in start_notify
    raise BleakError(f"Characteristic {char_specifier} not found!")
bleak.exc.BleakError: Characteristic 17 not found!
```
ble-deep scan result of daly BLE sensor as below
```
SERVICE 00001801-0000-1000-8000-00805f9b34fb (Handle: 10): Generic Attribute Profile
     CHARACTERISTIC 00002a05-0000-1000-8000-00805f9b34fb (Handle: 11): Service Changed ['read', 'indicate']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 13): Client Characteristic Configuration
SERVICE 0000fff0-0000-1000-8000-00805f9b34fb (Handle: 14): Vendor specific
     CHARACTERISTIC 0000fff1-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific ['read', 'notify']
         DESCRIPTOR 00002901-0000-1000-8000-00805f9b34fb (Handle: 18): Characteristic User Description
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 17): Client Characteristic Configuration
     CHARACTERISTIC 0000fff2-0000-1000-8000-00805f9b34fb (Handle: 19): Vendor specific ['read', 'write-without-response', 'write']
         DESCRIPTOR 00002901-0000-1000-8000-00805f9b34fb (Handle: 22): Characteristic User Description
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 21): Client Characteristic Configuration
SERVICE 02f00000-0000-0000-0000-00000000fe00 (Handle: 23): Unknown
     CHARACTERISTIC 02f00000-0000-0000-0000-00000000ff01 (Handle: 31): Unknown ['write-without-response', 'write']
     CHARACTERISTIC 02f00000-0000-0000-0000-00000000ff00 (Handle: 29): Unknown ['read']
     CHARACTERISTIC 02f00000-0000-0000-0000-00000000ff02 (Handle: 26): Unknown ['read', 'notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 28): Client Characteristic Configuration
     CHARACTERISTIC 02f00000-0000-0000-0000-00000000ff04 (Handle: 33): Unknown ['read', 'write-without-response', 'write', 'notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 35): Client Characteristic Configuration
     CHARACTERISTIC 02f00000-0000-0000-0000-00000000ff05 (Handle: 36): Unknown ['read', 'write-without-response', 'write']
     CHARACTERISTIC 02f00000-0000-0000-0000-00000000ff03 (Handle: 24): Unknown ['read']
```
Any directions or hints.

dreadnought, 2022-10-23:
Hi,
you can try to investigate it with bluetoothctl. Start `bluetoothctl` and run in its CLI `scan on`, `connect 26:62:04:12:34:56` (enter with your Daly devices MAC, that should be in the scan output if you're not already connected) and `info`. The output should look like this:
```
[DL-266204123456]# info
Device 26:62:04:12:34:56 (public)
	Name: DL-266204123456
	Alias: DL-266204123456
	Paired: no
	Trusted: no
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (0000fff0-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (f000ffc0-0451-4000-b000-000000000000)
	Modalias: usb:v045Ep0040d0300
	ManufacturerData Key: 0x6226
	ManufacturerData Value:
  04 03 00 75                                      ...u            
```
Check if your output looks similar, especially that the "Battery Service" is present. If not, double-check if you've really chosen the right MAC address. If you can't connect, ensure that you're not connected with your smartphone in parallel.

Which type/generation of Bluetooth module do you have? Does it look like this?
[image]

twhtanghk, 2022-10-26:
bluetoothctl as shown below:
```
	Name: DL-4016090151AD
	Alias: DL-4016090151AD
	Paired: no
	Bonded: no
	Trusted: no
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (0000fff0-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (02f00000-0000-0000-0000-00000000fe00)
	ManufacturerData Key: 0x0102
	ManufacturerData Value:
  04                                               .               
	RSSI: -87
```
No such battery service available. The outlook is similar to the one you posted above.

rokitar, 2022-11-09:
Hey,
I had almost the same problem. Then I started some reverse engineering following this tutorial:
https://www.hackster.io/news/david-tabachnikov-opens-up-a-proprietary-bluetooth-smart-light-with-a-little-reverse-engineering-0334e43e23f2

Finding out, that my "Smart BMS v.2.2.6" sends "d2030080003ed7b9" to the Daly Bluetooth Interface. The response is a 129 bit long message containing cell and total voltage, current and soc. And tons of "\x00". 

Hope this somehow helps. Seems to be a different protocol - maybe someone can integrate this into this great project. Thanks alot to dreadnought for the work!

sand1812, 2023-09-21:
Hi.
Same issue here. Last batch of BMS I bough has the same behaviour (I've used used dozen of daly bms via BT before, and I never had this issue).
Someone managed to solve this ?

LeaseOnLife, 2023-10-03:
adding myself to the list with this issue, got 2 fresh Daly Smart BMS 4S 100A with SW version 32_230627_001T

h2soft-de, 2023-10-05:
Hi. I found a workaround for the issue. I have an old bluetooth dongle and also the new one. I used LightBlue on Android and bluetoothctl on Linux. On LightBlue the old dongle is named "HI-Link-B20" and the new one "Keiser Corporation".
The UUIDs stays same but not the handles. So for new dongle you have to use:
handle 17 -> await self.client.start_notify('0000fff1-0000-1000-8000-00805f9b34fb', self._notification_callback) 
handle 48 (await self.client.write_gatt_char(48, bytearray(b""))) -> remove
handle 15 -> await self.client.write_gatt_char('0000fff2-0000-1000-8000-00805f9b34fb', value)

I used https://github.com/tomatensaus/python-daly-bms/blob/main/bin/daly-bmsBT-cli and now it works for old and new dongle.

LeaseOnLife, 2023-10-06:
> Hi. I found a workaround for the issue. I have an old bluetooth dongle and also the new one. I used LightBlue on Android and bluetoothctl on Linux. On LightBlue the old dongle is named "HI-Link-B20" and the new one "Keiser Corporation". The UUIDs stays same but not the handles. So for new dongle you have to use: handle 17 -> await self.client.start_notify('0000fff1-0000-1000-8000-00805f9b34fb', self._notification_callback) handle 48 (await self.client.write_gatt_char(48, bytearray(b""))) -> remove handle 15 -> await self.client.write_gatt_char('0000fff2-0000-1000-8000-00805f9b34fb', value)
> 
> I used https://github.com/tomatensaus/python-daly-bms/blob/main/bin/daly-bmsBT-cli and now it works for old and new dongle with 16s 250A BMS

Thanks much, that helps on my  BMS 4S 100A. At least I can read --soc and --status now and send / log that via MQTT. not all commands work yet, specifically --cell-voltages.

GeoTimber, 2023-10-16:
Yes thanks that works so far, but not cell voltages, this is what I get for --status

daly-bmsBT-cli --device=40:17:05:01:77:64 --all

returns

and for who is interesting I get the same results on a Daly Smart Active Balancer 1A  and a Daly Smart BMS 8s 24v 150A, so the balancer has results too.

Would really like to get the cell voltages, anyone who can help?

```
WARNING  [daly_bms_bluetooth.py:129] Timeout while waiting for 95 response
WARNING  [daly_bms_bluetooth.py:129] Timeout while waiting for 95 response
WARNING  [daly_bms_bluetooth.py:129] Timeout while waiting for 95 response
ERROR    [daly_bms_bluetooth.py:74] 95 failed after 3 tries
ERROR    [daly_bms.py:259] unkonwn status_field cells
WARNING  [daly_bms_bluetooth.py:129] Timeout while waiting for 96 response
WARNING  [daly_bms_bluetooth.py:129] Timeout while waiting for 96 response
WARNING  [daly_bms_bluetooth.py:129] Timeout while waiting for 96 response
ERROR    [daly_bms_bluetooth.py:74] 96 failed after 3 tries
ERROR    [daly_bms.py:259] unkonwn status_field temperature_sensors
```

```json
{
  "soc": {
    "total_voltage": 26.5,
    "current": -10.4,
    "soc_percent": 94.0
  },
  "cell_voltage_range": {
    "highest_voltage": 3.324,
    "highest_cell": 7,
    "lowest_voltage": 3.317,
    "lowest_cell": 8
  },
  "temperature_range": {
    "highest_temperature": 19,
    "highest_sensor": 1,
    "lowest_temperature": 19,
    "lowest_sensor": 1
  },
  "mosfet_status": {
    "mode": "discharging",
    "charging_mosfet": true,
    "discharging_mosfet": true,
    "capacity_ah": 263.2
  },
  "status": {
    "cells": 8,
    "temperature_sensors": 1,
    "charger_running": false,
    "load_running": false,
    "states": {
      "DI1": false
    },
    "cycles": 3
  },
  "cell_voltages": null,
  "temperatures": null,
  "balancing_status": {
    "1": false,
    "2": false,
    "3": false,
    "4": false,
    "5": false,
    "6": false,
    "7": false,
    "8": false
  },
  "errors": []
}`

GeoTimber, 2023-10-17:
Ok I worked out why the cell voltages did not return with the new dongle the amount of frames returned seems to be correct, I will try to to submit a pull request later this week or make a fork from tomatensaus for the new dongle, see this bit of code below, it tries to fix an inconsistency that is now solved

[image]

sand1812, 2023-11-15:
Hi guys,

Thanks a lot for you answers.

I've put everything together in my repo. 
https://github.com/sand1812/python-daly-bms.git

So you have :
- Old and new dongle working
- cells voltages ok
- set-soc for BT (this function is important for me, since I have few BMS with drifting SoC.)

I'll keep testing few more days (I have ~20 Daly BMS running, old & new dongle, or USB). If everything is ok, I'll do a pull request on dreadnought's repo.

dg1kwa, 2024-04-11:
> Hi guys,
>
> Thanks a lot for you answers.
>
> I've put everything together in my repo. https://github.com/sand1812/python-daly-bms.git
>
> So you have :
>
>     * Old and new dongle working
>
>     * cells [...]

I still become this:

python3 daly-bmsBT-cli -d 40:17:08:01:08:F3 --status

Traceback (most recent call last):
  File "/home/pi/daly-bms-test/python-daly-bms-main/bin/daly-bmsBT-cli", line 282, in <module>
    result = asyncio.run(bms.get_status())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pi/daly-bms-test/python-daly-bms-main/bin/daly-bmsBT-cli", line 31, in get_status
    await self.connect()
  File "/home/pi/daly-bms-test/python-daly-bms-main/bin/daly-bmsBT-cli", line 16, in connect
    self.connected = await self.bt_bms.connect(mac_address=self.mac_address)
  File "/usr/local/lib/python3.9/dist-packages/dalybms/daly_bms_bluetooth.py", line 45, in connect
    await self.client.start_notify(17, self._notification_callback)
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/__init__.py", line 822, in start_notify
    raise BleakError(f"Characteristic {char_specifier} not found!")
bleak.exc.BleakError: Characteristic 17 not found!

dg1kwa, 2024-04-11:
> Hi guys,
>
> Thanks a lot for you answers.
>
> I've put everything together in my repo. https://github.com/sand1812/python-daly-bms.git
>
> So you have :
>
>     * Old and new dongle working
>
>     * cells [...]

I try with python3 daly-bmsBT-cli --device 40:17:08:01:08:F3 --all --retry=1  but wait more then 1 hour four output allready :(

Daly BMS ist connected:

[DL-4017080108F3]# devices
Device 40:17:08:01:08:F3 DL-4017080108F3
[DL-4017080108F3]#

shaunol, 2024-08-24:
I had some issues with my new 16S Daly K Series Smart BMS with BT UART dongle which thanks to the above notes I was able learn the updated characteristic IDs but no combination of everything could get it to fully work for me. The main change that worked for me was for daly_bms_bluetooth.py was to use address=4 instead of address=8 - I'm not sure why this affects anything but I couldn't get anything working at all with address=8. Working through the read data functions one by one, I made the changes I wanted to make to get it all working for me. I didn't consider support for the different versions of BMS or USB dongle, other comments above from did take that into consideration. I used the example BT CLI from above just for quick testing.

You can see my changes here maybe it can help: https://github.com/shaunol/python-daly-bms/commit/655c91d955774b039655ff2c7fd7998c71cc53da

[image]

For troubleshooting the --verbose CLI flag is useful, to get the full logger output, I think now the biggest issue with connecting is not gracefully disconnecting when CTRL+C the CLI tool but that's fine. For later integration into an app I will have to be careful about this and also make sure it times out and retries connections in a reasonable way. Thanks everyone for your help and making this very useful tool.

hamadmad, 2025-06-14:
@shaunol's variant also works for my 13S "Smart Active Balancer" (the one with two UART ports). Thank you!