dreadnought/python-daly-bms #34: Error "UTC: Under Voltage in Charge" at low Temperature (sinowealth)

SamLue, 2022-12-13:
I would expect a message like "Charching temperature too low" but not "UTC: Under Voltage in Charge".
And what does "UTC: Under Voltage in Charge" mean?
I'm also wondering about "CHGMOS: Charging enabled". Shouldn't it be disabled at this temperatur?

Here the output:

`  "soc": {
    "total_voltage": 13.271,
    "current": -0.149,
    "soc_percent": 95.0
  },
  "mosfet_status": {
    "full_capacity_ah": 280.0,
    "remaining_capacity_ah": 267.36,
    "pack_state": [
      "FAST_DSG: Fast Discharging",
      "DSGING: Discharging",
      "DSGMOS: Discharging enabled",
      "CHGMOS: Charging enabled"
    ]
  },
  "status": {
    "cycles": 12
  },
  "cell_voltages": {
    "1": 3.315,
    "2": 3.327,
    "3": 3.32,
    "4": 3.312
  },
  "temperatures": {
    "external1": -8.4,
    "external2": 0.1
  },
  "errors": [
    "UTC: Under Voltage in Charge"
  ]
}
`

dreadnought, 2022-12-13:
Hi,
you're right, some of the error descriptions are wrong, I will correct them when I've time.

Have a look at page 5 of this PDF in the meantime:
[Sinowealth BMS Tool Introduction V0.1 English.pdf](https://github.com/dreadnought/python-daly-bms/files/10221258/Sinowealth.BMS.Tool.Introduction.V0.1.English.pdf)

SamLue, 2022-12-13:
OK,

thanks for your answer.
If only the description is wrong, maybe I can help to fix it.

I put the descriptions from the PDF in "Battery_Status"

`BATTERY_STATUS = {
        1: 'CTO: Disconnection protection occurs',
        2: 'AFE_SC: Hardware short circuit protection occurs',
        3: 'AFE_OV: Hardware overvoltage protection occurs',
        4: 'UTD: Discharge low temperature protection',
        5: 'UTC: Charge low temperature protection occurs',
        6: 'OTD: Discharge high temperature protection',
        7: 'OTC: Charge high temperature protection occurs',
        12: 'OCD: Discharge overcurrent protection occurs',
        13: 'OCC: Charge overcurrent protection occurs',
        14: 'UV: Undervoltage protection occurs',
        15: 'OV: Overvoltage protection occurs',
    }`

Do you want copy&paste this? Or should I do a pull request?

SamLue, 2022-12-13:
For better copy&paste not as code:

BATTERY_STATUS = {
        1: 'CTO: Disconnection protection occurs',
        2: 'AFE_SC: Hardware short circuit protection occurs',
        3: 'AFE_OV: Hardware overvoltage protection occurs',
        4: 'UTD: Discharge low temperature protection',
        5: 'UTC: Charge low temperature protection occurs',
        6: 'OTD: Discharge high temperature protection',
        7: 'OTC: Charge high temperature protection occurs',
        12: 'OCD: Discharge overcurrent protection occurs',
        13: 'OCC: Charge overcurrent protection occurs',
        14: 'UV: Undervoltage protection occurs',
        15: 'OV: Overvoltage protection occurs',
    }

commit 7a3cce7a (2022-12-14): Correct error messages of Sinowealth BMS (resolves #34)
closed 2022-12-14 by dreadnought