dreadnought/python-daly-bms #39: Feature: CAN Support

derlucas, 2023-01-17:
Hey,
because i want to use the Daly BMS via CAN i ordered a Makerbase CANable USB Stick and implemented your library with CAN.

I put the files here: https://gist.github.com/derlucas/d548562492db963dda240f63ed2877d6
The CAN Bus Specification can be found here: https://xleave.de/wiki/lib/exe/fetch.php?media=projekte:1_daly_can_protocol.pdf

Maybe someone can integrate my code properly into your Library, i don't have time unfortunately neither are my python skills that good :)

HappySamuel, 2023-09-11:
Hi @derlucas 

Appreciate for your effort to roll out CAN version of dalybms library. But what's the step to install it as library? Because when i run it, it's still using the default python-daly-bms library, instead of your CAN version.

Best,
Samuel

derlucas, 2023-09-11:
Hey Samuel,
i did not make a proper library nor did i make a pull request to get my code integrated into python-daly-bms. I just wanted to share my code in the hope someone may integrate it into the lib.
Up until then you have to fetch my code manually and integrate it into your project. Sorry for the misunderstanding.

Lucas