dreadnought/python-daly-bms #41: MQTT sees only first value

clertes, 2023-07-12:
Hi,

I'm using daly-bms-cli to send Daly BMS data to my mosquitto MQTT server which is then parsed via telegraf into influxdb.

I can see that daly-bms-cli is publishing the data when I start it with --verbose:

`daly-bms-cli -d /dev/ttyUSB0 --sinowealth --all --mqtt --mqtt-topic dalybms --mqtt-broker localhost --mqtt-user XXXXXXX --mqtt-password YYYYYYYYYYY --verbose
DEBUG    [client.py:2529] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x0b\x02'), 0a0b02
DEBUG    [daly_sinowealth.py:118] 337117 (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x10\x04'), 0a1004
DEBUG    [daly_sinowealth.py:118] fffff9e3b6 (5)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x13\x02'), 0a1302
DEBUG    [daly_sinowealth.py:118] 0021b1 (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x11\x04'), 0a1104
DEBUG    [daly_sinowealth.py:118] 000445c094 (5)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x12\x04'), 0a1204
DEBUG    [daly_sinowealth.py:118] 000173337f (5)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x15\x02'), 0a1502
DEBUG    [daly_sinowealth.py:118] 404b8f (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x14\x02'), 0a1402
DEBUG    [daly_sinowealth.py:118] 000721 (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x01\x02'), 0a0102
DEBUG    [daly_sinowealth.py:118] 0cb2f6 (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x02\x02'), 0a0202
DEBUG    [daly_sinowealth.py:118] 0cf102 (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x03\x02'), 0a0302
DEBUG    [daly_sinowealth.py:118] 0ce85b (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x04\x02'), 0a0402
DEBUG    [daly_sinowealth.py:118] 0cddb2 (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x05\x02'), 0a0502
DEBUG    [daly_sinowealth.py:118] 000045 (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x0c\x02'), 0a0c02
DEBUG    [daly_sinowealth.py:118] 0bd244 (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\r\x02'), 0a0d02
DEBUG    [daly_sinowealth.py:118] 0aab2f (3)
DEBUG    [daly_sinowealth.py:93] message: bytearray(b'\n\x16\x02'), 0a1602
DEBUG    [daly_sinowealth.py:118] 000018 (3)
DEBUG    [daly-bms-cli:131] Send data: 13.169 on topic: dalybms/soc/total_voltage, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m1), 'b'dalybms/soc/total_voltage'', ... (6 bytes)
DEBUG    [daly-bms-cli:131] Send data: -1.565 on topic: dalybms/soc/current, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m2), 'b'dalybms/soc/current'', ... (6 bytes)
DEBUG    [daly-bms-cli:131] Send data: 33.0 on topic: dalybms/soc/soc_percent, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m3), 'b'dalybms/soc/soc_percent'', ... (4 bytes)
DEBUG    [daly-bms-cli:131] Send data: 280.0 on topic: dalybms/mosfet_status/full_capacity_ah, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m4), 'b'dalybms/mosfet_status/full_capacity_ah'', ... (5 bytes)
DEBUG    [daly-bms-cli:131] Send data: 95.03 on topic: dalybms/mosfet_status/remaining_capacity_ah, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m5), 'b'dalybms/mosfet_status/remaining_capacity_ah'', ... (5 bytes)
DEBUG    [daly-bms-cli:131] Send data: ["FAST_DSG: Fast Discharging", "DSGING: Discharging", "DSGMOS: Discharging enabled", "CHGMOS: Charging enabled"] on topic: dalybms/mosfet_status/pack_state, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m6), 'b'dalybms/mosfet_status/pack_state'', ... (112 bytes)
DEBUG    [daly-bms-cli:131] Send data: 7 on topic: dalybms/status/cycles, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m7), 'b'dalybms/status/cycles'', ... (1 bytes)
DEBUG    [daly-bms-cli:131] Send data: 3.25 on topic: dalybms/cell_voltages/1, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m8), 'b'dalybms/cell_voltages/1'', ... (4 bytes)
DEBUG    [daly-bms-cli:131] Send data: 3.313 on topic: dalybms/cell_voltages/2, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m9), 'b'dalybms/cell_voltages/2'', ... (5 bytes)
DEBUG    [daly-bms-cli:131] Send data: 3.304 on topic: dalybms/cell_voltages/3, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m10), 'b'dalybms/cell_voltages/3'', ... (5 bytes)
DEBUG    [daly-bms-cli:131] Send data: 3.293 on topic: dalybms/cell_voltages/4, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m11), 'b'dalybms/cell_voltages/4'', ... (5 bytes)
DEBUG    [daly-bms-cli:131] Send data: 29.6 on topic: dalybms/temperatures/external1, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m12), 'b'dalybms/temperatures/external1'', ... (4 bytes)
DEBUG    [daly-bms-cli:131] Send data: 0.1 on topic: dalybms/temperatures/external2, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m13), 'b'dalybms/temperatures/external2'', ... (3 bytes)
DEBUG    [daly-bms-cli:131] Send data: [] on topic: dalybms/errors, retain flag: False
DEBUG    [client.py:2529] Sending PUBLISH (d0, q0, r0, m14), 'b'dalybms/errors'', ... (2 bytes)
DEBUG    [client.py:2529] Sending DISCONNECT`

Unfortunately I can only see the first value when I'm using chronograf/grafana, in this case it's `dalybms/soc/total_voltage`.

I've also tried to listen to the MQTT messages via "MQTT Explorer" where I can also confirm that it seems like only the first value is being sent.

My telegraf.conf is configured like this:

` [[inputs.mqtt_consumer]]
#   ## Broker URLs for the MQTT server or cluster.  To connect to multiple
#   ## clusters or standalone servers, use a separate plugin instance.
#   ##   example: servers = ["tcp://localhost:1883"]
#   ##            servers = ["ssl://localhost:1883"]
#   ##            servers = ["ws://localhost:1883"]
   servers = ["tcp://127.0.0.1:1883"]
   username = "XXXXXXX"
   password = "YYYYYYYYYYY"
#
#   ## Topics that will be subscribed to.
   topics = [
     "dalybms/#",
   ]

   data_format = "value"
   data_type = "float"`

As I'm running out of ideas, do I miss something?

sanitariu, 2023-09-19:
Did you fixed this ?

clertes, 2023-09-19:
Nope, went with another solution.

closed 2023-09-19 by clertes

sanitariu, 2023-09-19:
Can you share ? I am also trying to draw my data from Daly BMS to some kind of web interface ...