dreadnought/python-daly-bms #43: Bluetooth connection - which device ?

wimg, 2023-08-10:
I'm trying to set up a connection with the Daly BMS over Bluetooth. But I can't find any documentation regarding which -d parameter to use to do this.
Does anyone know the simple steps ?
I've seen some complex scripts from a different library, but those all require additional packages that pip can't seem to find...

dimikout3, 2023-09-08:
Hello! I am facing the same issue here. Does anyone have any relevant instruction to provide?

Marv085, 2023-09-23:
Hello, i have the same question, the BL connection is not possible. I tried to understand how the script is going from the "daly-bms-cli"  to "daly_bms_bluetooth.py" but i haven't found any connection. The cli command doesnt accept the mac_address as device and also it doesn't accept if the device is binded via "sudo rfcomm bind 0 XX:XX:XX:XX:XX:XX" as "/dev/rfcomm0". So it's not possible to connect via bluetooth by typing:
1. daly-bms-cli -d /dev/rfcomm0 --uart --soc
2.  daly-bms-cli -d XX:XX:XX:XX:XX:XX --uart --soc

dreadnought, 2023-11-16:
The CLI tool doesn't have the functionality to connect via Bluetooth implemented yet, only the Python module does. If you're interested in using the Python module to create a small tool for your purpose, check out the other [Bluetooth related](https://github.com/dreadnought/python-daly-bms/issues?q=is%3Aissue+label%3ABluetooth+) issues.