dreadnought/python-daly-bms #49: mqtt-hass not send any to the mqtt broker

byte4geek, 2024-04-29:
Hi,

i'm trying to send data to home assistant but when i run the script nothing are send to the mqtt broker, what i wrong?
This is the command i run:
```
root@dockers-deb12:~/daly/python-daly-bms# daly-bms-cli -d /dev/ttyUSB0 --soc --mqtt-hass --mqtt-broker 192.168.1.2 --mqtt-user mqttuser --mqtt-password xxxxxxx
{
  "total_voltage": 54.0,
  "current": 47.3,
  "soc_percent": 35.7
}```

byte4geek, 2024-04-29:
i run the mosquittopub on home assistant to see if the messages are received:

`mosquitto_sub -h 192.168.1.2 -d -v -u mqttuser -P xxxxxxx -t '#'`

mjrsnyder, 2025-02-21:
I think you're missing the `--mqtt` argument.

Try running it like this:
`daly-bms-cli -d /dev/ttyUSB0 --soc --mqtt --mqtt-hass --mqtt-broker 192.168.1.2 --mqtt-user mqttuser --mqtt-password xxxxxxx`